open62541/open62541 #2791: UA_server_browse max

etm, 2019-06-05:
## Description

UA_server_browse triggers weird behavior when creating a server. Consider the following function

```c
static bool node_get_reference(UA_Server *server, UA_NodeId parent, UA_NodeId *result) { 
  UA_BrowseDescription bDes;
  UA_BrowseDescription_init(&bDes);
  bDes.nodeId = parent;
  bDes.resultMask = UA_BROWSERESULTMASK_ALL;
  UA_BrowseResult bRes = UA_Server_browse(server, 1, &bDes);                                                                                                  

  if (bRes.referencesSize > 0) {
    UA_ReferenceDescription *ref = &(bRes.references[0]);

    *result = ref->nodeId.nodeId;
    UA_BrowseResult_deleteMembers(&bRes);
    UA_BrowseResult_clear(&bRes);
    return true;
  }
  return false;
}
```
When using this method in the following scenario:

```c
  for (i=0; i < 1000; i++) {
    node_get_reference(...);
    UA_Server_addObjectNode(....);
 }
```

After about 5 iterations through the loop, the server starts logging:

```
[2019-06-05 16:09:24.354 (UTC+0200)] info/session	Connection 0 | SecureChannel 0 | Session g=00000001-0000-0000-0000-000000000000 | AddNodes: Type for ns=2;s=... does not match node class
[2019-06-05 16:09:24.354 (UTC+0200)] info/session	Connection 0 | SecureChannel 0 | Session g=00000001-0000-0000-0000-000000000000 | AddNodes: Parent node ns=2;s=... not found
[2019-06-05 16:09:24.355 (UTC+0200)] info/session	Connection 0 | SecureChannel 0 | Session g=00000001-0000-0000-0000-000000000000 | AddNodes: The parent reference for ns=2;s=... is invalid with status code BadParentNodeIdInvalid
[2019-06-05 16:09:24.355 (UTC+0200)] info/session	Connection 0 | SecureChannel 0 | Session g=00000001-0000-0000-0000-000000000000 | AddNodes: Parent node ns=2;s=... not found
```

From this point on, no additional nodes can be created.

When increasing the maxRef parameter in UA_Server_browse to 2 it works for another five iterations, and so on. One coulde always use maxrefs with a VERY high number, but hmmmm :-)

Used CMake options:

-- Architectures included in amalgamation: posix
-- The selected architecture is: posix
-- Could NOT find Sphinx (missing: SPHINX_EXECUTABLE) 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/mangler/Projects/_/open62541/build
-- Cache values
BUILD_SHARED_LIBS:BOOL=ON
CLANG_FORMAT_EXE:FILEPATH=/usr/bin/clang-format
CMAKE_BUILD_TYPE:STRING=Debug
CMAKE_INSTALL_PREFIX:PATH=/usr/local
LIB_INSTALL_DIR:PATH=lib
UA_ARCHITECTURE:STRING=posix
UA_BUILD_EXAMPLES:BOOL=OFF
UA_BUILD_TOOLS:BOOL=OFF
UA_BUILD_UNIT_TESTS:BOOL=OFF
UA_ENABLE_AMALGAMATION:BOOL=ON
UA_ENABLE_COVERAGE:BOOL=OFF
UA_ENABLE_DISCOVERY:BOOL=ON
UA_ENABLE_DISCOVERY_MULTICAST:BOOL=OFF
UA_ENABLE_ENCRYPTION:BOOL=ON
UA_ENABLE_EXPERIMENTAL_HISTORIZING:BOOL=OFF
UA_ENABLE_HISTORIZING:BOOL=OFF
UA_ENABLE_METHODCALLS:BOOL=ON
UA_ENABLE_NODEMANAGEMENT:BOOL=ON
UA_ENABLE_QUERY:BOOL=OFF
UA_ENABLE_STATIC_ANALYZER:STRING=OFF
UA_ENABLE_SUBSCRIPTIONS:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS_EVENTS:BOOL=OFF
UA_LOGLEVEL:STRING=300
UA_NAMESPACE_ZERO:STRING=REDUCED
## Checklist

Please provide the following information:

 - [ e84eaf3] open62541 Version (release number or git tag):
 - [-] Other OPC UA SDKs used (client or server): 
 - [ Fedora 30] Operating system:
 - [see above ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [not necessary ] Wireshark network dump attached
 - [see above] Self-contained code example attached
 - [Sure] Critical issue

jpfr, 2019-06-05:
#2732 will land soon.
Let's see if the issue remains.

jpfr, 2019-06-05:
Hmm. This sounds bad.
What happens if you run the example with valgrind?

etm, 2019-06-05:
Valgrind looks fine.

==5802== HEAP SUMMARY:
==5802==     in use at exit: 87,795 bytes in 2,428 blocks
==5802==   total heap usage: 8,898 allocs, 6,470 frees, 353,031 bytes allocated
==5802== 
==5802== LEAK SUMMARY:
==5802==    definitely lost: 0 bytes in 0 blocks
==5802==    indirectly lost: 0 bytes in 0 blocks
==5802==      possibly lost: 0 bytes in 0 blocks
==5802==    still reachable: 87,795 bytes in 2,428 blocks
==5802==         suppressed: 0 bytes in 0 blocks
==5802== Rerun with --leak-check=full to see details of leaked memory
==5802== 
==5802== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)

etm, 2019-06-05:
@jpfr?

I tried for several hours to create a small example, without all the complexity my server has. 

Could the problem be that i use UA_Server_browse inside the callback function of a UA_Server_forEachChildNodeCall?

Maybe a commin results table with faulty management for nested cases?

jpfr, 2019-06-06:
> Could the problem be that i use UA_Server_browse inside the callback function of a UA_Server_forEachChildNodeCall?

If that is the issue, we have to fix it.
Browse should not modify the information model in any way.

The mentioned branch has merged to master.
Can you update and retry?

etm, 2019-07-08:
Sorry for the late response. The nature of the error changed somehow. 

When setting maxrefs to 1:

While before i got an error for each node i tried to insert, now i get the following error once.

Connection 0 | SecureChannel 0 | Session g=00000001-0000-0000-0000-000000000000 | AddNodes: Node type ns=42864;i=2696489728 not found

where the ns and the numeric id are random. The error message is always the second message that appears when starting the server. If i increase maxrefs (as before) the error vanishes. But at least all nodes are here now (or at least i cant see any missing).

When setting maxrefs to 2:
Error vanishes. Will try with a big server with 1000s of nodes tomorrow to be sure, to find out if it just occurs with higher node count.

teccheck, 2025-03-13:
Since there has been no more activity, I'll close this issue as outdated. If it's still relevant, feel free to reopen

closed 2025-03-13 by teccheck