open62541/open62541 #2792: Processing the message failed with error BadOutOfMemory

MarineKing5248, 2019-06-05:
## Description
I run the OPCUA python server on an ubuntu machine, and use another mac machine to run 1000 parallel processes to simulate 1000 clients to send message(an array of 100 doubles) to the server with 1 second time interval to do some test. It works well. Then I try to run the OPCUA C Server(with the same configuration) on the same ubuntu machine and use the same mac machine to run the clients to send message to the server. But now some thing happened, the first 44 connections between server and clients are totally fine, but the server will always stop the other connections( i.e. begins from the 45th connection). It shows the error message "Connection 45 | Processing the message failed with error BadOutOfMemory" on the ubuntu server and the error message "CRITICAL:opcua.client.ua_client.Socket:Received an error: MessageAbort(error:StatusCode(Good), reason:None)" on the mac clients' terminal.

## Background Information / Reproduction Steps
I double checked the background information, like: memory usage is quite enough( like 70% percent available). I changed the number of clients, as long as there are more than 44 clients, it will always stop the connections from the 45th client. 

I checked lots' of places but did find the solutions. Hope someone has any idea what happened or the possible solution of it. 

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag):
          Git-Revision:0.3-rc2
 - [x] Other OPC UA SDKs used (OPCUA python clients): 
         opcua==0.98.6
 - [x] Operating system:
         server--Ubuntu 18.04.2 LTS
         clients--macOs Mojave 10.14.5
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2019-06-06:
> CRITICAL:opcua.client.ua_client.Socket:Received an error: MessageAbort(error:StatusCode(Good), reason:None

That sounds like the problem is with python opcua.
Wrong forum. Please post at FreePython's github.

MarineKing5248, 2019-06-06:
> > CRITICAL:opcua.client.ua_client.Socket:Received an error: MessageAbort(error:StatusCode(Good), reason:None
> 
> That sounds like the problem is with python opcua.
> Wrong forum. Please post at FreePython's github.

But when I am using python server and python clients there is no error at all. It only happens when I am using open62541 server and python clients. That is why I think the problem is with the C server.

mlgiraud, 2019-06-07:
Could you check if you encounter the problem with the code in #2408 ?

mathierauf, 2020-02-19:
I have encountered a similiar problem: after more than 40 connections the server returned with a BADOUTOFMEMORY. 
It seems that even with security disabled (set to none-none) there needs to be a secure channel object created for every session. But while  'maxSessions' is set to 100, 'maxSecureChannels' is set to 40 in the in the default config. 
-> so increasing 'maxSecureChannels' solved the issue,  at least in my case...
Maybe it would make sense to set both variables to the same value in the default config?

kusumy, 2020-04-27:
Using open62541 on raspberry PI, i can achieve 1020 client connections. I thing the connection is not affected by time interval. I tried with 1 second and 5 second interval, i could not have more than 1020 connections. Although i have set maxSecureChannel and maxSession to 2000.

Wondering how can i increase the connections on open6541.

teccheck, 2025-03-13:
Since the new event loop landed, much of this has been overhauled and this limitation should not apply anymore

closed 2025-03-13 by teccheck