open62541/open62541 #2793: Can report.ci be helpful?

escherstair, 2019-06-06:
I've just found [report.ci](https://report.ci/) tool 

> Report.Ci takes you test results and generates a detailed test report on github, including annotations, so you know what went wrong.
> 
> Besides statistics, this includes annotations that will show up in pull request diffs.
> 

Do you think that open62541 project could somehow benefit from this tool?

Pro, 2019-06-07:
I personally like gui reporting tools. These help to understand what is going on.

Especially for out unit tests we would need some nicer output. Finding the error output in the currently over thousand lines is sometimes quite painful.

So yes, we could benefit from such a tool.
I looked through the supported unit test frameworks, but it seems that report.ci does not support the check framework (https://libcheck.github.io/check/) which we are using.

Correct?

escherstair, 2019-06-07:
I think you're right.

escherstair, 2019-06-07:
I've just asked the developer if he has plans to support Check framework too.

teccheck, 2025-03-13:
Thanks for the suggestion but report.ci seems to be down now

closed 2025-03-13 by teccheck