open62541/open62541 #2797: "BadSecurityChecksFailed" is received in UAExpert when handling client certificates

Nik654, 2019-06-07:
## Description
Hello,

We are using server from open62541 SDK version 0.3, following the [documentation](https://open62541.org/doc/0.3/), but regarding client certificates with have faced an issue.
## Background Information / Reproduction Steps
We are trying to set up and use trusted/revocation lists, and we are loading those lists from the filesystem.
UA_ENABLE_ENCRYPTION is enabled, and a server certificate is loaded, and passed through to the configuration method (UA_ServerConfig_new_basic256sha256).
We are loading certificate generated by UAExpert to trust list and same certificate (generated by UAExpert) is sent to open62541 server from UAExpert using OPC protocol.
When we try to connect, error "BadSecurityChecksFailed" is received in UAExpert, picture from UAExpert attached 
[image]

Do you have any suggestions on what approach we should use to further investigate this issue?

Used CMake options:

```bash
cmake -DUA_ENABLE_FULL_NS0=ON -DUA_ENABLE_ENCRYPTION=ON ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 0.3 release
 - [x] Other OPC UA SDKs used (client or server): UAExpert as OPC client
 - [x] Operating system: Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

mlgiraud, 2019-06-11:
Could you post the server logs?
Also please make sure you are using the correct certificate. It is located in `[uaexpert_dir]/PKI/own/certs/uaexpert.der`

SebsSa, 2019-06-14:
I just stumbled across this thread.
@Nik624 
If you don't find the path. Start UAExpert and click on Settings -> Manage Certificates -> Open Cer...

@mlgiraud
So if I start my server (UA_ServerConfig_new_basic256sha256) with : 
`./myServer server_cert.der server_key.der uaexpert.der`

Then it should look like this in UAExpert : 

[image]

Right? Now the question : Can ONLY the UAExpert access the server now?

mlgiraud, 2019-06-15:
Secure connections will only be accepted from uaexpert. The None endpoint can still be used by every client.
Keep in mind that this is not accesscontrol. Accesscontrol has a separate mechanism in OPC UA.

Nik654, 2019-06-18:
Hi,
We found cause of the problem, it was in our code, when loading certificates.
Anyway, thanks for suggestions.
I will close this issue.

closed 2019-06-18 by Nik654