open62541/open62541 #2799: Build fails as UA_NS0ID_HASINTERFACE cannot be found

cpipero, 2019-06-07:
## Description
just cloned the latest master. 
Ran cmake and then make as usual (I can build previous branches), but get stuck in the build at `ua_server_utils.c` with this error:
```ua_server_utils.c:168:33: error: ‘UA_NS0ID_HASINTERFACE’ undeclared here (not in a function)
     {0, UA_NODEIDTYPE_NUMERIC, {UA_NS0ID_HASINTERFACE}};```

I looked for the UA_NS0ID_HASINTERFACE string all over just in case (it's probably generated) and cannot find it. 

Here's the CMake options that I have ON:

```UA_BUILD_EXAMPLES:BOOL=ON
UA_BUILD_TOOLS:BOOL=ON
UA_BUILD_UNIT_TESTS:BOOL=ON
UA_ENABLE_DA:BOOL=ON
UA_ENABLE_DISCOVERY:BOOL=ON
UA_ENABLE_DISCOVERY_SEMAPHORE:BOOL=ON
UA_ENABLE_ENCRYPTION:BOOL=ON
UA_ENABLE_HARDENING:BOOL=ON
UA_ENABLE_METHODCALLS:BOOL=ON
UA_ENABLE_NODEMANAGEMENT:BOOL=ON
UA_ENABLE_NODESET_COMPILER_DESCRIPTIONS:BOOL=ON
UA_ENABLE_PUBSUB:BOOL=ON
UA_ENABLE_PUBSUB_DELTAFRAMES:BOOL=ON
UA_ENABLE_PUBSUB_INFORMATIONMODEL:BOOL=ON
UA_ENABLE_PUBSUB_INFORMATIONMODEL_METHODS:BOOL=ON
UA_ENABLE_STATUSCODE_DESCRIPTIONS:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS:BOOL=ON
UA_ENABLE_TYPENAMES:BOOL=ON
UA_MSVC_FORCE_STATIC_CRT:BOOL=ON```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Debian 9.6 Stretch 
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Pro, 2019-06-07:
Did you also update the submodules?

```
cd open62541
git submodule sync
git submodule update --init --recursive
```

cpipero, 2019-06-07:
Thank you Stefan, that did it. 
What happened is that I actually downloaded the .zip (instead of clonining) and manually copied the ua nodeset artifacts in the deps folder. Clearly that is not a good practice (although it worked in the past).

From now on I will stick with the right process:
```
git clone https://github.com/open62541/open62541.git
cd open62541
git submodule sync
git submodule update --init --recursive
mkdir build
cd build
cmake -LA ..
make
```

closed 2019-06-07 by cpipero