open62541/open62541 #2800: PubSub Publisher example, not setting the DataSetMetaData

cpipero, 2019-06-07:
## Description
I'm running the ` tutorial_pubsub_publish.c ` example and everything works fine, as long as I sue the sample code for the subscriber.

But in truth, a true PubSub publisher should also inform potential susbcribers/clients/configurators of its configuration. That's why we have the PubSub information model and methods.

**Take as an example** the .NET reference implementation from the OPC Foundation, and look at the DataSet configuration in its address space.

[image]

And look at the same property using the open62541 example:

[image]

This prevents those 3rd parties to interact with my publisher. 

## Testing attempts
I have been poking around the issues and the test code and the best I could some up was to try code the DataSetMetaData using the availabel facilities, but so far I have no luck. I don't get any error, but the property does not get set.

Here the complete example with the added Metadata code in the ` addPublishedDataSet() ` function:

```
/* This work is licensed under a Creative Commons CCZero 1.0 Universal License.
 * See http://creativecommons.org/publicdomain/zero/1.0/ for more information. */

/**
 * .. _pubsub-tutorial:
 *
 * Working with Publish/Subscribe
 * ------------------------------
 *
 * Work in progress: This Tutorial will be continuously extended during the next
 * PubSub batches. More details about the PubSub extension and corresponding
 * open62541 API are located here: :ref:`pubsub`.
 *
 * Publishing Fields
 * ^^^^^^^^^^^^^^^^^
 * The PubSub publish example demonstrate the simplest way to publish
 * informations from the information model over UDP multicast using the UADP
 * encoding.
 *
 * **Connection handling**
 *
 * PubSubConnections can be created and deleted on runtime. More details about
 * the system preconfiguration and connection can be found in
 * ``tutorial_pubsub_connection.c``.
 */

#include <open62541/plugin/log_stdout.h>
#include <open62541/plugin/pubsub_ethernet.h>
#include <open62541/plugin/pubsub_udp.h>
#include <open62541/server.h>
#include <open62541/server_config_default.h>
#include <ua_pubsub.h>
#include <signal.h>
#include <open62541/server_pubsub.h>
#include <open62541/types.h>

UA_NodeId connectionIdent, publishedDataSetIdent, writerGroupIdent, dataSetWriterIdent;

UA_PublishedDataSetConfig publishedDataSetConfig;
UA_DataSetWriterConfig dataSetWriterConfig;

static void
addPubSubConnection(UA_Server *server, UA_String *transportProfile,
                    UA_NetworkAddressUrlDataType *networkAddressUrl){
    /* Details about the connection configuration and handling are located
     * in the pubsub connection tutorial */
    UA_PubSubConnectionConfig connectionConfig;
    memset(&connectionConfig, 0, sizeof(connectionConfig));
    connectionConfig.name = UA_STRING("Demo UADP Connection");
[... 228 lines cut ...]
UA_ENABLE_PUBSUB_INFORMATIONMODEL_METHODS:BOOL=ON
UA_ENABLE_QUERY:BOOL=OFF
UA_ENABLE_STATIC_ANALYZER:STRING=OFF
UA_ENABLE_STATUSCODE_DESCRIPTIONS:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS_EVENTS:BOOL=OFF
UA_ENABLE_TYPENAMES:BOOL=ON
UA_ENABLE_UNIT_TESTS_MEMCHECK:BOOL=OFF
UA_ENABLE_UNIT_TEST_FAILURE_HOOKS:BOOL=OFF
UA_ENABLE_VALGRIND_INTERACTIVE:BOOL=OFF
UA_FILE_NS0:BOOL=OFF
UA_LOGLEVEL:STRING=300
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 0.4.0
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Debian 9.6 Stretch
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [x] Critical issue: trying to get it working for the OPAF Interop next week ;-)

jpfr, 2019-06-08:
@andreasebner Can you comment on this?

mpostol, 2019-06-08:
@cpipero said:

`But in truth, a true PubSub publisher should also **inform potential susbcribers/clients/configurators** of its configuration. That's why we have the PubSub information model and methods.'

Let me recall a well-known definition:

### Publisher-subscriber communication pattern

Publish-subscribe is a messages distribution scenario where senders of messages, called publishers, do not send them directly to specific receivers, called subscribers, but instead categorize published messages into classes without knowledge of which subscribers if any, there may be. Similarly, subscribers express interest in one or more classes and only receive messages that are of interest, without knowledge of which publishers, if any, there are. In the publish-subscribe model, subscribers typically receive only a subset of the total messages published. The process of selecting messages for reception and processing is called filtering. There are two common forms of filtering: topic-based and content-based.

Read the full story [OPC UA PubSub Main Technology Features](https://github.com/mpostol/OPC-UA-OOI/blob/Networking/Networking/SemanticData/README.PubSubMTF.md)

I hope you may be also interested to read:  [Underlying Transport over Ethernet](https://github.com/mpostol/OPC-UA-OOI/blob/Networking/Networking/SemanticData/README_EthernetMapping.md)

Hope it helps.
Mariusz

jpfr, 2019-06-09:
Fixed in #2802. Can you confirm @cpipero ?

cpipero, 2019-06-09:
Thank you @jpfr . We are getting closer.
I added the changes and yes, this improves things.

I now have to understand if there is anything I should add in my code.

Let me explain details (sorry for the lengthy description, but I hope it helps).

## Adding the Fix
I added the changes in #2802 . That basically exposes the ` PublishedData ` information correctly **_and without me needing to change the code_** in the ` tutorial_pubsub_publish.c ` file (it basically just works, as I'd expect).

Here's a screenshot:

[image]

However, the ` DataSetMetaData ` property of the DataSet is not announced correctly (_**but at least is not null anymore**_).

In this screenshot you see the comparison between the .NET server and the tutorial_pubsub_publish, where the ` Fields ` array is empty where it should instead list the one DataField for the current time:

[image]

## 
## Additional Improvements Needed
This part of the PubSub information model is important because it gives the potential subscribers the needed info to map the DataSetWriter's fields.

Look at the .NET Sample Config UI that now can successfully navigate to the remote writer dataset (now it doesn' t crash). But if I want to configure the DataReader, I am missing the necessary metadata:

[image]

## 
## Next Steps
Maybe I need to create the UA_DataSetMetaDataType object myself, but I would wonder why the PublishedData get populated automatically (which is what changed in the ` src/pubsub/ua_pubsub_ns0.c  ` file).

I will try a few things while you read this.

Thank you again for the tremendous help.

cpipero, 2019-06-10:
## Small Update

Upon more investigation of the following code (copied from above, repeated for convenience)

```c
[...]
case UA_NS0ID_PUBLISHEDDATAITEMSTYPE: {
            UA_PublishedDataSet *publishedDataSet = UA_PublishedDataSet_findPDSbyId(server, *myNodeId);
            if(!publishedDataSet)
                return;
            switch(nodeContext->elementClassiefier) {
                case UA_NS0ID_PUBLISHEDDATAITEMSTYPE_PUBLISHEDDATA: {
                    UA_PublishedVariableDataType *pvd = (UA_PublishedVariableDataType *)
                            UA_calloc(publishedDataSet->fieldSize, sizeof(UA_PublishedVariableDataType));
                    size_t counter = 0;
                    UA_DataSetField *field;
                    LIST_FOREACH(field, &publishedDataSet->fields, listEntry) {
                        pvd[counter].attributeId = UA_ATTRIBUTEID_VALUE;
                        pvd[counter].publishedVariable = field->config.field.variable.publishParameters.publishedVariable;
                        //UA_NodeId_copy(&field->config.field.variable.publishParameters.publishedVariable, &pvd[counter].publishedVariable);
                        counter++;
                    }
                    UA_Variant_setArray(&value, pvd, publishedDataSet->fieldSize,
                                        &UA_TYPES[UA_TYPES_PUBLISHEDVARIABLEDATATYPE]);
                    break;
                }
[...]
```

the ` publishedDataSet->fields `  do not contain any metadata also. 

This is something that I think the UA_PubSubManager should do...

I'm now looking  at the ` ua_pubsub_manager.c `  trying to make sense of it

cpipero, 2019-06-12:
@jpfr would you have any idea on this? It might make it for a great Interop week if I can fix this.

If you look at my last comment, I'm either very close or way off track.

Thanks in advance,

Cos

jpfr, 2019-06-12:
Hey Cos,

you are looking at the right code to set the content during reading.
Two things are required:

1. Set and store the `UA_FieldMetaData` in the internal DataSet representation
2. Return the fieldmetadata in the `onRead` callback.

The `UA_DataSetField` structure (`publishedDataSet->fields`) contains a `UA_FieldMetaData` entry.
So you can copy that over during onRead.

But the `UA_FieldMetaData` is not automatically filled atm.
Most of the information can be read from the variable definition when the field is created.

We might have to extend the public API so that the user can fill in more MetaData information.
At this time you need to expose some internals to add the data for your demo.

andreasebner, 2019-06-12:
@cpipero thanks for your feedback and detailed reports. As you already recognized, the pubsub meta data are currently not supported. The DataSetMetaData should be created and updated by the pubsub core without any configuration.

One short comment to your example code: The OPC UA PubSub Standard defines four methods to add published informations (AddPublishedDataItems 9.1.4.5.2., AddPublishedDataItemsTemplate 9.1.4.5.4., AddPublishedEvents 9.1.4.5.3. and AddPublishedEventsTemplate 9.1.4.5.5). The so called 'Template' methods allow to inject a complete pubsub configuration into the server including the metadata. The non template methods create the metadata (if supported) on the fly. The union 'itemsTemplate' within the UA_PublishedDataSetConfig is used to provide the template functionalty by the open62541 PubSub API. If the 'standard' PublishedDataItems are used you will not need to configure the 'itemTemplate'.

We are currently working on a PubSub feature list, which will contain supported/partly supportet and currently not supported aspects of the pubsub standard. I assume the finialization and release of this list next week. I agree with your statement about the importance of the PubSub-metadata for the subscribers. We will work on this feature and try to provide a first implementation as soon as possible.

cpipero, 2019-06-14:
Thank you guys,
I have been looking at a way to "cheat" just to do some final testing at the interop this week.

In reality, as you are probably saying, things should start happening when the server adds datafields.

But in the meanwhile, I tried to modify the _OnRead_ call in the ` ua_pubsub_ns0.c `.

I'm not successful yet, because I don't have much time left and I'm just shooting in the dark.

The following code (very ugly and should NEVER be user) does succeed on the server side, but the client doesn't like it and eventually leads to bad outcomes.

But I put it here for you to look at and see if it can be easily "steered" in the right direction.

I'm making up the field name (using the tutorial sample I will only have one field all the times).

Aagin, this is just for you to see if modifying the _OnRead_ this way makes sense. Clearly it needs to be improved.

```c
static void
onRead(UA_Server *server, const UA_NodeId *sessionId, void *sessionContext,
       const UA_NodeId *nodeid, void *context,
       const UA_NumericRange *range, const UA_DataValue *data) {
    UA_Variant value;
    UA_Variant_init(&value);
    const UA_NodePropertyContext *nodeContext = (const UA_NodePropertyContext*)context;
    const UA_NodeId *myNodeId = &nodeContext->parentNodeId;

    switch(nodeContext->parentClassifier){
    case UA_NS0ID_PUBSUBCONNECTIONTYPE: {
        UA_PubSubConnection *pubSubConnection =
            UA_PubSubConnection_findConnectionbyId(server, *myNodeId);
        switch(nodeContext->elementClassiefier) {
        case UA_NS0ID_PUBSUBCONNECTIONTYPE_PUBLISHERID:
            if(pubSubConnection->config->publisherIdType == UA_PUBSUB_PUBLISHERID_STRING) {
                UA_Variant_setScalar(&value, &pubSubConnection->config->publisherId.numeric,
                                     &UA_TYPES[UA_TYPES_STRING]);
            } else {
                UA_Variant_setScalar(&value, &pubSubConnection->config->publisherId.numeric,
                                     &UA_TYPES[UA_TYPES_UINT32]);
            }
            break;
        default:
            UA_LOG_WARNING(&server->config.logger, UA_LOGCATEGORY_SERVER,
                           "Read error! Unknown property.");
        }
        break;
    }
    case UA_NS0ID_WRITERGROUPTYPE: {
        UA_WriterGroup *writerGroup = UA_WriterGroup_findWGbyId(server, *myNodeId);
        if(!writerGroup)
            return;
        switch(nodeContext->elementClassiefier){
        case UA_NS0ID_WRITERGROUPTYPE_PUBLISHINGINTERVAL:
            UA_Variant_setScalar(&value, &writerGroup->config.publishingInterval,
                                 &UA_TYPES[UA_TYPES_DURATION]);
            break;
        default:
            UA_LOG_WARNING(&server->config.logger, UA_LOGCATEGORY_SERVER,
                           "Read error! Unknown property.");
        }
        break;
    }
    case UA_NS0ID_PUBLISHEDDATAITEMSTYPE: {
        UA_PublishedDataSet *publishedDataSet = UA_PublishedDataSet_findPDSbyId(server, *myNodeId);
        if(!publishedDataSet)
            return;
        switch(nodeContext->elementClassiefier) {
        case UA_NS0ID_PUBLISHEDDATAITEMSTYPE_PUBLISHEDDATA: {
            UA_PublishedVariableDataType *pvd = (UA_PublishedVariableDataType *)
                UA_calloc(publishedDataSet->fieldSize, sizeof(UA_PublishedVariableDataType));
[... 33 lines cut ...]
            UA_Variant_setScalarCopy(&value, dsmd, &UA_TYPES[UA_TYPES_DATASETMETADATATYPE]);

            break;
        }
        case UA_NS0ID_PUBLISHEDDATAITEMSTYPE_CONFIGURATIONVERSION: {
            UA_Variant_setScalarCopy(&value, &publishedDataSet->dataSetMetaData.configurationVersion,
                                     &UA_TYPES[UA_TYPES_CONFIGURATIONVERSIONDATATYPE]);
            break;
        }
        default:
            UA_LOG_WARNING(&server->config.logger, UA_LOGCATEGORY_SERVER,
                           "Read error! Unknown property.");
        }
        break;
    }
    default:
        UA_LOG_WARNING(&server->config.logger, UA_LOGCATEGORY_SERVER,
                       "Read error! Unknown parent element.");
    }
    UA_StatusCode retval = UA_Server_writeValue(server, *nodeid, value);
    if (retval != UA_STATUSCODE_GOOD) UA_LOG_ERROR(&server->config.logger, UA_LOGCATEGORY_SERVER, "Read error! %s", UA_StatusCode_name(retval));
}
```

Sorry I am doing this in a rush. I will look at it more this weekend with more tme.

jpfr, 2019-06-14:
> UA_realloc(dsmd->fields, dsmd->fieldsSize);

This is not big enough.
It should be `UA_realloc(dsmd->fields, dsmd->fieldsSize * sizeof(UA_FieldMetaData))`;

jpfr, 2019-06-14:
And then you can just say `UA_FieldMetaData_init(&dsmd->fields[i]);` to init.

cpipero, 2019-06-14:
Thanks @jpfr  , that was a silly one...
And I forgot that I had an example in the subscriber tutorial. I edited my code above. 
Still no decoding but I'm getting closer. 

I know I'm implementing all this one level above where it should be but it helps understanding. We probably need to add some good documentation on this whole pub-sub. Let me know if I can help

cpipero, 2019-09-03:
Just capturing a note from our release meeting. Cross reference Pull Request #3052 that starts fixing this.

Stasik0, 2021-05-18:
Bump in 2021. Looks like the DataSetMetaData of ReaderGroup1 is still empty...
[image]

o6-nbeier, 2026-10-20:
Thank you for taking the time to open this issue and describe it so clearly.
The problem has since been fixed on the maintained branches, so we are closing it as resolved.

closed 2026-10-20 by o6-nbeier