open62541/open62541 #2803: Trustlist / What to pass

SebsSa, 2019-06-11:
## Description

I am looking for a documentation about the trustlist.
Or a help how procedure or create a trust list and to provide the file as an input to application?

## Background Information / Reproduction Steps

Hello,
I have a question about the trustlist. I'm trying to get a server running that can ONLY be called/used by my local UAExpert. 

Currently I am using the :

- server_encryption.c (example)
- server_cert.der server_key.der (create_self-signed.py)

After doing some research, I found out that 

> trust lists are .pem files with the certificates you trust.

So I opened UAExpert.  And found this here in the documentary

> because the Server has to trust the Client’s certificate as well ...  UaExpert’s certificate from pki/rejected to ....

And here I have a problem: Actually the server should reject me and the document should be created. But if the TrustList is empty the Server will accept every Client Certificate.

Now I've taken the only existing .pem-file I found ,
/.../.config/unifiedautomation/uaexpert/PKI/own/uaexpert_key.pem

and called:
`./serverOne server_cert.der server_key.der uaexpert_key.pem `

but nothing happens after the call. No error ! No warning ! No Server 

Can somebody give me a few tips?

Related questions:
#1708 #1554 #2205

Please provide the following information:

 - [x] open62541 Version (release number or git tag): latest version
 - [x] Other OPC UA SDKs used (client or server): UAExpert
 - [x] Operating system: Debian
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

mlgiraud, 2019-06-11:
Supplying the key as a certificate in the revocation list is certainly wrong. You have to supply the certificate of uaexpert. If you want to create a trust list that consists of only the uaexpert certificate it suffices to just load the uaexpert.der certificate as the trustlist. You can find this in `[uaexpert_dir]/PKI/own/certs/uaexpert.der`

SebsSa, 2019-06-11:
Ah sure ! My bad. 
Could you give me one more hint about the warning : (

`There has to be at least one endpoint.`

mlgiraud, 2019-06-11:
Most probably there was an error during initialization of the security policies. You could try to debug into the code.

SebsSa, 2019-06-12:
The answer referred to an edited question from above: 

Could you give me one more hint about the warning :

`There has to be at least one endpoint.`

Thanks again

SebsSa, 2019-06-12:
Okay, I'm gonna have to reopen it. Sorry

Currently I'm using the server_basic128rsa15 script (it works exactly as it should).
**Added the.answer and UA_UsernamePasswordLogin**

After this started to put the UAExpert client in the trust list. As suggested above.

`./myServer server_cert.der server_key.der certstore/uaexpert.der`

Throws:
>  [2019-06-12 15:45:11.672 (UTC+0200)] info/network	TCP network layer listening on opc.tcp://Moros:4840/

Then I wanted to verify that only the UAExpert client can access the server and quickly used the client.c I found in the examples.

Changed :  
  
`retval = UA_Client_connect_username(client, "opc.tcp://localhost:4840", "admin", "password");`
to
`retval = UA_Client_connect_username(client, "opc.tcp://Moros:4840", "newAdmin", "newPassword");`

And : 
`sudo gcc -std=c99 -lmbedtls -lmbedx509 -lmbedcrypto client.c open62541.c open62541.h -o myClient`
`./myClient `

Throws:

> [2019-06-12 15:34:06.873 (UTC+0200)] info/client	SecurityPolicy not specified -> use default #None
> [2019-06-12 15:34:06.873 (UTC+0200)] warn/securitypolicy	No PKI plugin set. Accepting all certificates
> [2019-06-12 15:34:06.874 (UTC+0200)] info/client	TCP connection established

and the client was able to search and read/write my nodes.

How is it possible that this client has access to my server?

SebsSa, 2019-06-12:
Okay, found my answer myself. 
(But I leave the question open, maybe someone is just as ~~dumb~~ overworked as I am)
My certificates were in the wrong directory!

But still getting this error :
[2019-06-12 16:06:01.895 (UTC+0200)] fatal/userland	Could not create the server config
Someone here has an idea?

My certificates were in the wrong directory

closed 2019-06-14 by SebsSa