open62541/open62541 #2805: UA_STATUSCODE_BADIDENTITYTOKENINVALID Returned by Service_ActivateSession() when using other client SDK's

cpipero, 2019-06-11:
## Description
In the OPAF Test, Schneider is using the Unified Automation High Performance SDK to write a client application. 
During testing, the 2 companies that use open62541 failed to connect (that includes me of course).

During debugging it boils down to an error thrown at the ` Service_ActivateSession() `  function in the ` ua_service_session.c `  file of the open62541 SDK.
It reached the point where it just can't match an endpoint and it returns a _Bad Token_ error.
```
/* No matching endpoint found */
    if(!ed) {
        response->responseHeader.serviceResult = UA_STATUSCODE_BADIDENTITYTOKENINVALID;
        return;
    }
```

Looking at wireshark, it appears that the the client request (performed with no encryption and no authentication) present itself without a _UserToken_ (not even anonymous). 
Chances are the server doesn't like that.

The other server SDK's do not seem to mind that and let the client in anyhow. I might have to change the code to allow for that too, but need more investigation.

## Background Information / Reproduction Steps
I can replicate this problem easily by building the sample clients using the Unified Automation evaluation SDK's (I did it with both their HDK ad their C++)

## Used CMake options:

```
UA_ENABLE_AMALGAMATION:BOOL=OFF
UA_ENABLE_COVERAGE:BOOL=OFF
UA_ENABLE_CUSTOM_NODESTORE:BOOL=OFF
UA_ENABLE_DA:BOOL=ON
UA_ENABLE_DETERMINISTIC_RNG:BOOL=OFF
UA_ENABLE_DISCOVERY:BOOL=ON
UA_ENABLE_DISCOVERY_MULTICAST:BOOL=OFF
UA_ENABLE_DISCOVERY_SEMAPHORE:BOOL=ON
UA_ENABLE_ENCRYPTION:BOOL=ON
UA_ENABLE_EXPERIMENTAL_HISTORIZING:BOOL=OFF
UA_ENABLE_HARDENING:BOOL=ON
UA_ENABLE_HISTORIZING:BOOL=OFF
UA_ENABLE_IMMUTABLE_NODES:BOOL=OFF
UA_ENABLE_JSON_ENCODING:BOOL=OFF
UA_ENABLE_MALLOC_SINGLETON:BOOL=OFF
UA_ENABLE_METHODCALLS:BOOL=ON
UA_ENABLE_MULTITHREADING:BOOL=OFF
UA_ENABLE_NODEMANAGEMENT:BOOL=ON
UA_ENABLE_NODESET_COMPILER_DESCRIPTIONS:BOOL=ON
UA_ENABLE_PUBSUB:BOOL=ON
UA_ENABLE_PUBSUB_CUSTOM_PUBLISH_HANDLING:BOOL=OFF
UA_ENABLE_PUBSUB_DELTAFRAMES:BOOL=ON
UA_ENABLE_PUBSUB_ETH_UADP:BOOL=OFF
UA_ENABLE_PUBSUB_INFORMATIONMODEL:BOOL=ON
UA_ENABLE_PUBSUB_INFORMATIONMODEL_METHODS:BOOL=ON
UA_ENABLE_QUERY:BOOL=OFF
UA_ENABLE_STATIC_ANALYZER:STRING=OFF
UA_ENABLE_STATUSCODE_DESCRIPTIONS:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS_EVENTS:BOOL=OFF
UA_ENABLE_TYPENAMES:BOOL=ON
UA_ENABLE_UNIT_TESTS_MEMCHECK:BOOL=OFF
UA_ENABLE_UNIT_TEST_FAILURE_HOOKS:BOOL=OFF
UA_ENABLE_VALGRIND_INTERACTIVE:BOOL=OFF
UA_FILE_NS0:BOOL=OFF
UA_LOGLEVEL:STRING=300
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 0.4.0
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Debian 9.6 Stretch
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [x] Critical issue: trying to get it working for the OPAF Interop this week ;-)

cpipero, 2019-06-11:
This is what I'm getting now 

```
[2019-06-11 11:15:54.575 (UTC-0400)] warn/channel	Connection 5 | SecureChannel 1 | Service request 473 without a valid session
[2019-06-11 11:15:54.575 (UTC-0400)] debug/server	Sent ServiceFault with error code BadSessionIdInvalid
[2019-06-11 11:15:54.626 (UTC-0400)] info/channel	Connection 5 | SecureChannel 1 | CloseSecureChannel
[2019-06-11 11:15:54.678 (UTC-0400)] info/network	Connection 5 | Closed
[2019-06-11 11:15:59.632 (UTC-0400)] info/network	Connection 5 | New connection over TCP from 127.0.0.1
[2019-06-11 11:15:59.695 (UTC-0400)] info/channel	Creating a new SecureChannel
[2019-06-11 11:15:59.695 (UTC-0400)] warn/securitypolicy	No PKI plugin set. Accepting all certificates
[2019-06-11 11:15:59.695 (UTC-0400)] info/channel	Connection 5 | SecureChannel 2 | Opened SecureChannel
[2019-06-11 11:15:59.745 (UTC-0400)] debug/channel	Connection 5 | SecureChannel 2 | Trying to create session
[2019-06-11 11:15:59.745 (UTC-0400)] info/channel	Connection 5 | SecureChannel 2 | Session a3f7fa40-e193-e87f-6426-12e1f5dfcd3c created
[2019-06-11 11:15:59.794 (UTC-0400)] debug/session	Connection 5 | SecureChannel 2 | Session ns=1;g=a3f7fa40-e193-e87f-6426-12e1f5dfcd3c | Execute ActivateSession
```

jpfr, 2019-06-11:
Thanks. Your fix is merged in #2806.
This is looking okay now.

> Service request 473 without a valid session
That means the client wants to delete a session that does not exist.
Logging was improved a bit here. But the code looks good.
If this is a problem on our end, it would be best to see a Wireshark dump.

closed 2019-06-12 by Pro