open62541/open62541 #2807: No definition for alloca in VxWorks

kingalexsylvester, 2019-06-12:
I am trying to compile open62541 current master branch in the VxWorks and I facing definition issue with alloca as it is not available in VxWorks OS. Just to get the binaries I have made it to malloc to make the stack compile.

But what is the permanent fix here? Is this topic already discussed?

## Checklist

Please provide the following information:

 - [*] open62541 Version (release number or git tag):534ca8b  
 - [*] Other OPC UA SDKs used (client or server): None
 - [*] Operating system: VxWorks Windriver
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [*] Critical issue

jpfr, 2019-06-12:
Depending on the architecture, open62541 uses alloca or variable-length arrays (VLA) on the stack.

It sounds you should switch from alloca to vla here:
https://github.com/open62541/open62541/blob/master/include/open62541/architecture_functions.h#L50

Note that the Linux kernel has managed to get rid of VLA entirely.
We could achieve the same. But it is work nobody has done so far.
https://lwn.net/Articles/749064/

kingalexsylvester, 2019-06-12:
Hello Julius,

Thanks for the reply and I shall get back to you with the results in this thread and we can close it so that other VxWorks Windriver users can make use of it during the porting for their targets.

Pro, 2019-06-12:
Maybe it is enough to add the folowing line to the ua_architecture.h function of VxWorks:
https://github.com/open62541/open62541/blob/master/arch/vxworks/ua_architecture.h

```
#  define UA_STACKARRAY(TYPE, NAME, SIZE) TYPE NAME[SIZE]
```
That will override the one in the generic header.

Does that fix it?

kingalexsylvester, 2019-06-16:
No need any additions and the definitions in architecture_functions.h itself was used.

closed 2019-06-16 by kingalexsylvester