open62541/open62541 #2808: CustomDataTypes based on other CustomDataTypes are not supported

TorbenD, 2019-06-12:
Hi,

using customDataTypes working fine as long as they are build of UA_TYPES. If they are build of other customDataTypes the server crashes because of following code:

open62541/ua_types.c lines 937-969
```
static UA_StatusCode
copyStructure(const void *src, void *dst, const UA_DataType *type) {
    UA_StatusCode retval = UA_STATUSCODE_GOOD;
    uintptr_t ptrs = (uintptr_t)src;
    uintptr_t ptrd = (uintptr_t)dst;
-->    const UA_DataType *typelists[2] = { UA_TYPES, &type[-type->typeIndex] };
    for(size_t i = 0; i < type->membersSize; ++i) {
        const UA_DataTypeMember *m= &type->members[i];
-->        const UA_DataType *mt = &typelists[!m->namespaceZero][m->memberTypeIndex];
        if(!m->isArray) {
            ptrs += m->padding;
            ptrd += m->padding;
            retval |= copyJumpTable[mt->typeKind]((const void*)ptrs, (void*)ptrd, mt);
            ptrs += mt->memSize;
            ptrd += mt->memSize;
        } else {
            ptrs += m->padding;
            ptrd += m->padding;
            size_t *dst_size = (size_t*)ptrd;
            const size_t size = *((const size_t*)ptrs);
            ptrs += sizeof(size_t);
            ptrd += sizeof(size_t);
            retval |= UA_Array_copy(*(void* const*)ptrs, size, (void**)ptrd, mt);
            if(retval == UA_STATUSCODE_GOOD)
                *dst_size = size;
            else
                *dst_size = 0;
            ptrs += sizeof(void*);
            ptrd += sizeof(void*);
        }
    }
    return retval;
}
```
In the first pointed line the typelist is just filled with the standard data types UA_TYPES and the customDataType. But the customDataTypes for the members are not listed. In the second line the memberTypeIndex is not working because it is out of the range.

We are using the generated files for datatypes and have to delete double dataTypes. We have to do this, because we want to have dataTypes only ones in one library and not in each. This is necessary because of transformation staff and we do not want to duplicate source code.

Do you have any suggestions how to solve this problem?

Thank you

jpfr, 2019-06-12:
Custom DataTypes can use other custom DataTypes as long as they end up in the same "UA_DataTypeArray" (see `open62541/types.h`).
The indexing is relative to the current datatype.

If this does not work, then this is a bug that needs to be solved.

Yeah, the docs need improvement here.

TorbenD, 2019-06-12:
Thanks for the answer. 

The problem is that if the DataTypes are in both UA_DataTypeArray you get different decoding and encoding functions, but they both have the same name:

```
/* IdEnum */
static UA_INLINE size_t
UA_IdEnum_calcSizeBinary(const UA_IdEnum *src) {
    return UA_calcSizeBinary(src, &UA_LIFECYCLEENTRY[UA_LIFECYCLEENTRY_IDENUM]);
}
static UA_INLINE UA_StatusCode
UA_IdEnum_encodeBinary(const UA_IdEnum *src, UA_Byte **bufPos, const UA_Byte *bufEnd) {
    return UA_encodeBinary(src, &UA_LIFECYCLEENTRY[UA_LIFECYCLEENTRY_IDENUM], bufPos, &bufEnd, NULL, NULL);
}
static UA_INLINE UA_StatusCode
UA_IdEnum_decodeBinary(const UA_ByteString *src, size_t *offset, UA_IdEnum *dst) {
    return UA_decodeBinary(src, offset, dst, &UA_LIFECYCLEENTRY[UA_LIFECYCLEENTRY_IDENUM], NULL);
}

/* Identification */
static UA_INLINE size_t
UA_Identification_calcSizeBinary(const UA_Identification *src) {
    return UA_calcSizeBinary(src, &UA_LIFECYCLEENTRY[UA_LIFECYCLEENTRY_IDENTIFICATION]);
}
static UA_INLINE UA_StatusCode
UA_Identification_encodeBinary(const UA_Identification *src, UA_Byte **bufPos, const UA_Byte *bufEnd) {
    return UA_encodeBinary(src, &UA_LIFECYCLEENTRY[UA_LIFECYCLEENTRY_IDENTIFICATION], bufPos, &bufEnd, NULL, NULL);
}
static UA_INLINE UA_StatusCode
UA_Identification_decodeBinary(const UA_ByteString *src, size_t *offset, UA_Identification *dst) {
    return UA_decodeBinary(src, offset, dst, &UA_LIFECYCLEENTRY[UA_LIFECYCLEENTRY_IDENTIFICATION], NULL);
}
```

```
/* IdEnum */
static UA_INLINE size_t
UA_IdEnum_calcSizeBinary(const UA_IdEnum *src) {
    return UA_calcSizeBinary(src, &UA_IDENTIFICATION[UA_IDENTIFICATION_IDENUM]);
}
static UA_INLINE UA_StatusCode
UA_IdEnum_encodeBinary(const UA_IdEnum *src, UA_Byte **bufPos, const UA_Byte *bufEnd) {
    return UA_encodeBinary(src, &UA_IDENTIFICATION[UA_IDENTIFICATION_IDENUM], bufPos, &bufEnd, NULL, NULL);
}
static UA_INLINE UA_StatusCode
UA_IdEnum_decodeBinary(const UA_ByteString *src, size_t *offset, UA_IdEnum *dst) {
    return UA_decodeBinary(src, offset, dst, &UA_IDENTIFICATION[UA_IDENTIFICATION_IDENUM], NULL);
}

/* Identification */
static UA_INLINE size_t
UA_Identification_calcSizeBinary(const UA_Identification *src) {
    return UA_calcSizeBinary(src, &UA_IDENTIFICATION[UA_IDENTIFICATION_IDENTIFICATION]);
}
static UA_INLINE UA_StatusCode
UA_Identification_encodeBinary(const UA_Identification *src, UA_Byte **bufPos, const UA_Byte *bufEnd) {
    return UA_encodeBinary(src, &UA_IDENTIFICATION[UA_IDENTIFICATION_IDENTIFICATION], bufPos, &bufEnd, NULL, NULL);
}
static UA_INLINE UA_StatusCode
UA_Identification_decodeBinary(const UA_ByteString *src, size_t *offset, UA_Identification *dst) {
    return UA_decodeBinary(src, offset, dst, &UA_IDENTIFICATION[UA_IDENTIFICATION_IDENTIFICATION], NULL);
}
```

jpfr, 2019-06-12:
> If the DataTypes are in both UA_DataTypeArray

You mean "are both in the same UA_DataTypeArray"?

> you get different decoding and encoding functions, but they both have the same name

You should not have to define DataTypes twice.
Normally, the datatypes the dependency is

- Between the builtin types
- From a custom type to a builtin type
- Between custom types in the same .bsd file

These cases are automatically covered.

Do you have a dependency between DataTypes from different .bsd files?

TorbenD, 2019-06-12:
Thats correct, but we need to make it more modular. With our nodeswitch extension [https://github.com/open62541/open62541/pull/2748](url) it is possible to load different nodestore at runtime and they all get different namespaces. So we want to seperate the dataTypes to the namespaces they belong to. And then it does not make sense to put all dataTypes in one namespace.

jpfr, 2019-06-13:
I adjusted the title to reflect the issue.

You receive the DataTypeArray together with the namespace definitions.
You could do a **little bit of preprocessing** before adding the DataTypeArray to the linked list in the server config.

For example by looking up the dependencies (via the NodeId of the type) and ensuring that DataType definitions that depend on one another end up in the same DataTypeArray.
E.g. by copying the relevant DataTypes to the current DataTypeArray.

That way we keep the encoding layer really simple and fast.
And the encoding layer can be "standalone" without dependencies to the Nodestore layer.

TorbenD, 2019-06-13:
That is a good idea and works if you have just dependencies between two data types.

The problem is that we have one basic customDataType and two more that both are using this but have no dependencies to each other. Than the problem with the names for the functions occurs again.

jpfr, 2019-06-13:
Assume you have three bsd files: bsd1, bsd2, bsd3
And bsd2 and bsd3 define the same datatype that relies on a type from bsd1.

- bsd1: type_a
- bsd2: type_b (relies on type_a)
- bsd3: type_b (relies on type_a)

Does this describe the situation?

TorbenD, 2019-06-13:
Yes that's right, but bsd2 is type_b and bsd3 is type_c

TorbenD, 2019-07-05:
Are there any considerations how to solve this problem?

teccheck, 2025-03-13:
This should work in current releases, so I'll close this issue