open62541/open62541 #2810: NamespaceArray looks good but Namespaces object doesn't match it

cpipero, 2019-06-12:
## Description

The ` NamespaceArray ` node contains all the added namespaces and it looks good, but the complete infomration should be added to the ` Namespaces ` object, which is a collection of  ` NamespaceMetaDataType ` objects which provide all sort of information about the configuration.

## Background Information / Reproduction Steps
This can be replicated easily by running any of the sample server applications.
Here's the look of it.

[image]

The server functions correctly, but compliance tests would fail. I have not tried the ` server_ctt ` sample yet.

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 0.4.0
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Debian 9.6
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

cpipero, 2019-06-16:
I tried the ` server_ctt ` example (and the ` server_nodeset ` and some of the tutorials too for good measure) and things are even worse: the 1 Namespaces ` object is actually _missing_ .

[image]

In the application we wrote, the ` Namespaces ` object does appears -- albeit missing the correct content. That's probably because we used the nodeset geenration python scripts that make explict calls to ` UA_Server_addNamespace() ` in the ` ua_server.c ` file.

For some reason, I can't replicate the same behavio using the ` tutorial_server_datasource.c ` example which makes the came calls.

## Note on Compliance
As far as the specification goes (Part 5, Section 6.3.1, Table 9),  the _Server_ object lists the _Namespaces_ component as _optional_ .
However, this component is much needed by the so called _aggregating servers_ which are getting some traction these days (the one I can think of is the one that Microsoft uses for their AzureIoT stuff, and of course, we wrote our own as well).

jpfr, 2019-06-16:
Yes, so far the Namespaces object is not "live".
It may be generated for a build with "full" information model if the object instance is defined explicitly in Nodeset2.xml. But then it will be just the hull of an object that does not change as namespaces are added.

Contributors are invited to chip in here.
If nobody steps up, this looks like a feature that has to wait until we are done with the "lower", i.e. embedded, server profiles.

cpipero, 2019-06-16:
Thank you @jpfr  .   I can look at this if you need (I don't know what means to be a "contributor" but I will be messing with this code for  a while). 

I would prefer that _Namespaces_ get added to/removed from together with the _NamespaceArray_ updates (granted the condition that NS_0 -- maybe "FULL" -- is enabled).

I'll take a crack at it and let you know. 
Thanks again.

jpfr, 2019-06-16:
Perfect. Looking forward to it. 👍