open62541/open62541 #2811: How to enable sequence number for data set messages in pubsub code?

SKShah36, 2019-06-12:
## Description

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

merlingeorge555, 2019-06-15:
Hello @SKShah36,

Question 1: I want to receive the sequence number of DataSetMessages in the subscriber part of the publish-subscribe tutorial. The DataSetMessages are generated by the DataSetWriter in the publisher application. However, I couldn't find a way to enable 'dataSetMessageSequenceNrEnabled'.
How you ensure that publisher is sending the datasetmessage sequence number?

>> The tutorial publisher application in the master branch does not send the datasetmessagesequence number in the network message. Have you verified this in the network trace?

>> If not, please do modify the code as below and then check the wireshark, Now you can see the datasetmessagesequence number is been sent from the publisher. To ensure you run the subscriber_standalone you can see the message length will be increased/

>> In src/pubsub/ua_pubsub.c, search UA_DataSetWriter_generateDataSetMessage() and change the default uadp configuration in line number:1953 as i.e add the UA_UADPDATASETMESSAGECONTENTMASK_SEQUENCENUMBER in the default configuration as below.

```
defaultUadpConfiguration.dataSetMessageContentMask = (UA_UadpDataSetMessageContentMask)
            ((u64)UA_UADPDATASETMESSAGECONTENTMASK_TIMESTAMP | (u64)UA_UADPDATASETMESSAGECONTENTMASK_MAJORVERSION |
             (u64)UA_UADPDATASETMESSAGECONTENTMASK_MINORVERSION | (u64) **UA_UADPDATASETMESSAGECONTENTMASK_SEQUENCENUMBER**);
```

>> And also, May I know the purpose why you running the pubsub_subscribestandalone? because now the subscriber target variable is implemented and networkmessage decoding part is already done within the stack. I would suggest you to run tutorial_pubsub_subscriber.c 

>> While running tutorial_pubsub_subscriber.c you can print and see the datasetmessagesequencenumber within the  UA_Server_DataSetReader_process() in src/pubsub/ua_pubsub.c line:1064 

[image]

[image]

Regards,
Merlin

SKShah36, 2019-06-17:
Thank you very much. It worked. t was indeed not publishing the sequence number. I am actually using the tutorial_pubsub_subscriber.c. Sorry for the confusion. I have updated the description.

Now I have a similar problem with enabling publisher id. This can become a separate question but since it is on similar lines I am posting it as a comment. Let me know if you would think I should post it as a separate one. I would happily close the issue.

 - Changed the writer group configuration
I understand that the writer group configuration consists of message settings and the message settings has a networkMessageContentMask that can be configured to contain publisher id

The following is what I've done:
``` 
static void
addWriterGroup(UA_Server *server) {
    /* Now we create a new WriterGroupConfig and add the group to the existing
     * PubSubConnection. */
    UA_WriterGroupConfig writerGroupConfig;
    memset(&writerGroupConfig, 0, sizeof(UA_WriterGroupConfig));
    writerGroupConfig.name = UA_STRING("Demo WriterGroup");
    writerGroupConfig.publishingInterval = publish_interval;
    writerGroupConfig.enabled = UA_TRUE;
    writerGroupConfig.writerGroupId = 100;
    writerGroupConfig.encodingMimeType = UA_PUBSUB_ENCODING_UADP;

    //MessageSettings for writer group

    UA_UadpWriterGroupMessageDataType defaultUadpConfiguration;
   //defaultUadpConfiguration.groupVersion = 12;
    defaultUadpConfiguration.networkMessageContentMask = UA_UADPNETWORKMESSAGECONTENTMASK_PUBLISHERID;
    memset(&defaultUadpConfiguration, 0, sizeof(UA_UadpWriterGroupMessageDataType));

    UA_ExtensionObject msgSettings;
    UA_ExtensionObject_init(&msgSettings);
    msgSettings.content.decoded.data = &defaultUadpConfiguration;
    writerGroupConfig.messageSettings = msgSettings;

            /* The configuration flags for the messages are encapsulated inside the
             * message- and transport settings extension objects. These extension
             * objects are defined by the standard. e.g.
             * UadpWriterGroupMessageDataType */
    UA_Server_addWriterGroup(server, connectionIdent, &writerGroupConfig, &writerGroupIdent);
}
```
Thanks again

merlingeorge555, 2019-06-17:
Hello @SkShah36,

That's cool. Can you please post in separate issue thread and close this issue? 

It will be more easier for me and as well as help others to track. 

Thanks,
Merlin

SKShah36, 2019-06-17:
Thanks @merlingeorge555. I am closing the issue after the positive response. Here's the link to the new issue: https://github.com/open62541/open62541/issues/2825#

closed 2019-06-17 by SKShah36