open62541/open62541 #2813: Issues with ExpandedNodeId

xallaume, 2019-06-13:
## Description
Hello. I've been struggling several days on this one. I create a very simple server with the following code, so as to obtain an ExpandedNodeId. No complain on execution, no valgrind issue. Then I run this simple client. Again, no complain and no valgrind issue. But I can never have the correct expandedNodeId.nodeId.namespaceIndex. It's always 0, whatever I try.

Suprisingly, when I use UA Expert to read the node value, I get the correct values for all ExpandedNodeId attributes. But if I try to update it from UA Expert, it doesn't complain and always returns me a expandedNodeId.nodeId.namespaceIndex=0. The other attributes are always correctly updating.

By other attributes of ExpandedNodeId, I mean the nodeId string identifier, the server index and the namespace uri.

# Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 0.3
 - [x] Other OPC UA SDKs used (client or server): UA Expert 1.5.1
 - [ ] Operating system: Ubuntu 18.10
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [ ] Critical issue

Code for the server :
```c
#include <signal.h> 
#include "open62541.h" 
UA_Boolean running = true; 
void signalHandler(int sig) 
{ running = false; } 

int main(int argc, char argv) { 
        signal(SIGINT, signalHandler); /* catch ctrl-c */ 
        /* Create a server listening on port 4840 */
        UA_ServerConfig *config = UA_ServerConfig_new_default();
        UA_Server *server = UA_Server_new(config);
        /* Add a variable node */
        /* Define the node attributes */
        UA_VariableAttributes attr = UA_VariableAttributes_default;
        UA_ExpandedNodeId myExpandedNodeId;
        UA_ExpandedNodeId_init( &myExpandedNodeId );
        myExpandedNodeId = UA_EXPANDEDNODEID_STRING( 1, "TestStringNodeId" );
        myExpandedNodeId.namespaceUri = UA_STRING( "Fake uri" );
        myExpandedNodeId.serverIndex = 4;
        UA_Variant_setScalar(&attr.value, &myExpandedNodeId, &UA_TYPES[UA_TYPES_EXPANDEDNODEID]);
        attr.description = UA_LOCALIZEDTEXT("en-US", "the answer");
        attr.displayName = UA_LOCALIZEDTEXT("en-US", "the answer");
        attr.dataType = UA_TYPES[UA_TYPES_EXPANDEDNODEID].typeId;
        attr.accessLevel = UA_ACCESSLEVELMASK_READ | UA_ACCESSLEVELMASK_WRITE;
        /* Define where the node shall be added with which browsename */
        UA_NodeId newNodeId = UA_NODEID_STRING(1, "the.answer");
        UA_NodeId parentNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);
        UA_NodeId parentReferenceNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES);
        UA_NodeId variableType = UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE);
        UA_QualifiedName browseName = UA_QUALIFIEDNAME(1, "the answer");
        /* Add the node */
        UA_StatusCode ret = UA_Server_addVariableNode(server, newNodeId, parentNodeId, parentReferenceNodeId,
                browseName, variableType, attr, NULL, NULL);
        /* Run the server loop */
        UA_StatusCode status = UA_Server_run(server, &running);
        UA_Server_delete(server);
        UA_ServerConfig_delete(config);
        return status;
}
```

Code for the client:
```c
#include <stdio.h>
#include "open62541.h"
int main(int argc, char *argv[])
{
        /* Create a client and connect */
        UA_Client *client = UA_Client_new(UA_ClientConfig_default);
        UA_StatusCode status = UA_Client_connect(client, "opc.tcp://localhost:4840");
        if(status != UA_STATUSCODE_GOOD) {
                UA_Client_delete(client);
                return status;
        }
        /* Read the value attribute of the node. UA_Client_readValueAttribute is a
        * wrapper for the raw read service available as UA_Client_Service_read. */
        UA_Variant value; /* Variants can hold scalar values and arrays of any type */
        UA_Variant_init(&value);
        status = UA_Client_readValueAttribute(client, UA_NODEID_STRING(1, "the.answer"), &value);
        if(status == UA_STATUSCODE_GOOD && UA_Variant_hasScalarType(&value, &UA_TYPES[UA_TYPES_EXPANDEDNODEID])) {
                printf("the value is: %i\n", (*(UA_ExpandedNodeId*)value.data).nodeId.namespaceIndex);
        }
        /* Clean up */
        UA_Variant_deleteMembers(&value);
        UA_Client_delete(client); /* Disconnects the client internally */
        return status;
}
```

Any hint, please?

jpfr, 2019-06-13:
Do you set a value to the NamespaceUri?

The standard says:
> The NamespaceUri is optional. If it is specified, then the NamespaceIndex inside the NodeId shall be ignored.

xallaume, 2019-06-13:
Now I feel stupid... But thank you, really!!!

closed 2019-06-13 by xallaume