open62541/open62541 #2814: Monitored Item - exception-based model [Question]

rudoq32, 2019-06-13:
## Description
In OP UA Part 4 [p. 60] 5.12.1.2 Sampling interval they say that _The Server may support data that is collected based on a sampling model or generated based on an **exception-based model**. The fastest supported sampling interval may be equal to 0, which indicates that the data item is exception-based rather than being sampled at some period. An exception-based model means that the underlying system does not require sampling and reports data changes._

Does it mean that my monitored item has to be Event based type, or I can achieve it by setting _samplingInterval_ to zero? For sure, once the server supports it. 

Because my server supports samplingInterval = 0 (at least I allow that limit), but once I tries to add monitoring item with  samplingInterval  = 0 I am getting BadInternalError.

Thanks Rudo

## Background Information / Reproduction Steps
// Configuration for monitoring
UA_MonitoringMode monitoringMode = UA_MONITORINGMODE_REPORTING;
UA_TimestampsToReturn timestampsToReturn = UA_TIMESTAMPSTORETURN_BOTH;
UA_Double samplingInterval = 0;
UA_Int32 queueSize = 0;
// Configuration for subscription
UA_Double requestedPublishingInterval = 1;

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=REDUCED -DUA_ENABLE_AMALGAMATION -DUA_ENABLE_SUBSCRIPTIONS -DUA_ENABLE_SUBSCRIPTIONS_EVENTS -DUA_ARCHITECTURE_FREERTOSLWIP -DOPEN62541_FEERTOS_USE_OWN_MEM
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag):0.3-rc2-1010-g456a58f3
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: FREERTOS
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2019-06-13:
That's a good point. So userland should be able to trigger variablechange notifications.

We then need to decide how and with which context the trigger is done

- Trigger the MonitoredItem
  - We need a way to register MonitoredItems (created by a client) in userland
  - E.g. with a callback where new MonitoredItems are announced

- Trigger the VariableNode
   - The Variable needs a linked list of the interested MonitoredItems (which it does not have at the moment)
   - A new public API method is created to manually trigger notifications to be created for a variable

jpfr, 2021-09-23:
Datachange MonitoredItem without a sampling interval have been added to master.
Those can also be used locally.

closed 2021-09-23 by jpfr