open62541/open62541 #2816: How to create a certification and send this to server

AndyarcherReset, 2019-06-14:
Hi,
Now I want to create a UA client and then connect to a third part UA server with the Basic128Rsa15 and Basic256.
But seems that do not support this, right?

SebsSa, 2019-06-14:
Right now, I'm working the other way around.
I want to generate a server and run it with Basic128Rsa15 and only allow a specific third party to use it. (UAExpert in my case)

That's why I'm not an expert, but maybe I can help you anyway. 

Read this through for yourself: 
https://github.com/open62541/open62541/issues/2637

and maybe use :
https://github.com/open62541/open62541/blob/master/tools/certs/create_self-signed.py  and
https://github.com/open62541/open62541/blob/master/examples/encryption/client_encryption.c

jpfr, 2019-06-14:
This is definitely supported.

See here the example:
https://github.com/open62541/open62541/blob/master/examples/encryption/client_encryption.c

You can define which SecurityPolicy to use in the client config:
https://open62541.org/doc/current/client.html#client-configuration

AndyarcherReset, 2019-06-17:
> This is definitely supported.
> 
> See here the example:
> https://github.com/open62541/open62541/blob/master/examples/encryption/client_encryption.c
> 
> You can define which SecurityPolicy to use in the client config:
> https://open62541.org/doc/current/client.html#client-configuration

Hi,
From the example client_encryption.c, I just see that can load the certificate.
But how can I create a certificate within the code? Or should I use another way to do this by myself?

jpfr, 2019-06-17:
https://github.com/open62541/open62541/blob/master/tools/certs/create_self-signed.py

PR #4535 (2021-08-09): [Review] feat(sec): Create certificate in C using OpenSSL/LibreSSL
Create a self-signed encryption certificate in C (only implemented with OpenSSL/LibreSSL).

For demonstration purpose, `UA_CreateCertificate` is implemented as a fallback in `server_encryption.c`, if no files are passed using the command line.

Resolves: #2816
closed 2021-08-09 by jpfr