open62541/open62541 #2817: Problem when using multiple DataSetReader

Fischer-Thomas, 2019-06-14:
## Description
I followed the pull request #2613 and tried the implementation myself.
However after the merge many things don't work anymore.
For example when working with 1 ReaderGroup and 2 DataSetReader the subscribed variables wont receive any data.
https://github.com/open62541/open62541/blob/534ca8b4026c1481db9fc67039e282bc99cbff86/src/pubsub/ua_pubsub.c#L400-L406

I located the problem in this code which only allows a single DataSetReader when there is only 1 ReaderGroup. So everytime the stack tries to process the message it doesn't get any DataSetReader.

Also there is no checking for the publisherId AND writerId in this scenario! There might occur errors/problems when there are two ore more publisher with a different config!

jpfr, 2019-06-14:
@opcua-tsn-team-kalycito Please have a look.

These things should be covered by unit tests to prevent regression in the future.

opcua-tsn-team-kalycito, 2019-06-18:
> ## Description
>
> I followed the pull request #2613 and tried the implementation myself.
> However after the merge many things don't work anymore.
> For example when working with 1 ReaderGroup and 2 [...]

@Fischer-Thomas, thank you for your interest in the subscriber implementation. Can you please use #2631 instead of #2613 as #2631 is the commit that was actually merged. Even in that case, our priority was to provide a quick implementation that focuses specifically on writing the subscribed value into the information model - this was previously not available and we are getting the ball rolling here for interested users. Additional items (including the ones you have mentioned) are definitely in our roadmap and we will be happy to let you know as soon as we have them and their dependencies in place. Please stay tuned for regular updates.

Regards,
opc-ua-tsn-team-kalycito

opcua-tsn-team-kalycito, 2019-08-21:
Hi @Fischer-Thomas,
We have addressed this issue in PR #2944.
The same has been merged to the master.

Regards
opcua-tsn-team-kalycito

closed 2021-04-07 by andreasebner