open62541/open62541 #2818: Publish/Subscribe nodeId

mrburen121, 2019-06-14:
## Description
I would like to pusblish (new) data through a server application. I'm trying to set a custom node number for each created object.

## Background Information / Reproduction Steps
The current problem that I'm facing is that the given node ID's are not set as active node ID as shown in UaExpert. Does anyone know what I'm doing wrong? I would like to set for weigher1 nodeID = 50220., weigher2 nodeID =  50224.

Reference project: [https://bitbucket.org/tlichten/pubsub-with-open62541/src/master/PubSubIssue/](url)
Reference code: [https://open62541.org/doc/current/tutorial_server_object.html](url)
[https://open62541.org/doc/current/tutorial_server_variabletype.html](url)
[https://open62541.org/doc/current/tutorial_server_object.html](url)

My static int code:
```
static int run(UA_String *transportProfile,
               UA_NetworkAddressUrlDataType *networkAddressUrl) {
    signal(SIGINT, stopHandler);
    signal(SIGTERM, stopHandler);

/* ---------- SAMPLE SERVE CODE -------------------------------------------------*/
    UA_Server *server = UA_Server_new();
	UA_ServerConfig *config = UA_Server_getConfig(server);
	UA_ServerConfig_setDefault(config);

	/* Details about the connection configuration and handling are located in
	 * the pubsub connection tutorial */
	config->pubsubTransportLayers =
		(UA_PubSubTransportLayer *) UA_calloc(2, sizeof(UA_PubSubTransportLayer));
	if(!config->pubsubTransportLayers) {
		UA_Server_delete(server);
		return EXIT_FAILURE;
	}
	config->pubsubTransportLayers[0] = UA_PubSubTransportLayerUDPMP();
	config->pubsubTransportLayersSize++;
#ifdef UA_ENABLE_PUBSUB_ETH_UADP
	config->pubsubTransportLayers[1] = UA_PubSubTransportLayerEthernet();
	config->pubsubTransportLayersSize++;
#endif
/* ---------- END SAMPLE SERVE CODE ----------------------------------------------*/

/* ---------- WEIGHER DEMO CODE --------------------------------------------------*/

	defineObjectTypes(server);

    // Add weigher objects
    addWeigherObjectInstance(server, "weigher1", 1, 2001);		// 1 = namespace ID, 2001 = Identifier
    addWeigherTypeConstructor(server);
    addWeigherObjectInstance(server, "weigher2", 1, 2002);
    addWeigherTypeConstructor(server);

	// Add Value Callback Methods
    addValueCallback(server, 1, 50220); 						// weigher1
	addValueCallback(server, 1, 50224); 						// weigher2

	// Publish and subscripbe
	addPubSubConnection(server, transportProfile, networkAddressUrl);
	addPublishedDataSet(server);
	addDataSetField(server);								// publish DateTime
	addWeigherDataSetField(server, 1, 50220, "weigher1");		// publish weigher1
	addWeigherDataSetField(server, 1, 50224, "weigher2");		// publish weigher2
	/* ---------- END WEIGHER SERVE CODE -----------------------------------------*/

	addWriterGroup(server);
        addDataSetWriter(server);

        UA_StatusCode retval = UA_Server_run(server, &running);

	UA_Server_delete(server);
	return retval == UA_STATUSCODE_GOOD ? EXIT_SUCCESS : EXIT_FAILURE;
}
```
UaExpert creenshot:
[image]

Git repo: [https://bitbucket.org/mrburen121/open62541_server/src/master/](url)

Used CMake options:
```bash
cmake -DBUILD_SHARED_LIBS=OFF -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_NAMESPACE_ZERO=FULL -UA_ENABLE_PUBSUB=ON -UA_ENABLE_PUBSUB_ETH_UADP=ON -UA_BUILD_EXAMPLES=OFF  -UA_ENABLE_SUBSCRIPTIONS=ON -UA_ENABLE_PUBSUB_INFORMATIONMODEL=ON   -UA_ENABLE_AMALGAMATION=ON ..
```

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): 0.4.0
 - [X] Other OPC UA SDKs used (client or server): UaExpert
 - [X] Operating system: Ubuntu 19.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [X] Self-contained code example attached
 - [ ] Critical issue

johndzilva, 2019-06-18:
@mrburen121 If you want to have your own nodeid (weigher1 nodeID = 50220) then you need to add the mandatory variables inside your addWeigherObjectInstance() function or else it will take generate random id for initialization.

UA_Server_addNode_begin();
      Call the mandatory variables with your custom node id specified
UA_Server_addNode_finish();

mrburen121, 2019-06-18:
Thank you! This was exactly where I was looking for.

closed 2019-06-18 by mrburen121