open62541/open62541 #2819: open62541 stack C99 standard issue in creating a VIP using VxWorks Windriver

kingalexsylvester, 2019-06-15:
I am not able to compile open62541 stack as VIP using VxWorks Windriver. It's the problem that Windriver is not using C99 by default. Have anybody come across this and fixed it. Please help.

jpfr, 2019-06-15:
To quote from their marketing material:

> The Wind River Compiler uses the Edison Design Group’s compiler front end to ensure full compliance with the most recent ANSI/ISO C and C++ compiler standards including the following:
> - ISO/IEC 9899:1990 (“C89”) standard
> - ISO/IEC 9899:1999 (“C99”) standard
> - ISO/IEC 14882:2003(E) C++ standard

So you should be able to switch settings to C99.

kingalexsylvester, 2019-06-16:
Adding **-std=c99** in the build spec did the job. I had to remove certain modules and do changes specific to VxWorks. I shall review the code and share a pull request if I feel its a necessary upgrade to the stack.

To mention I did not use CMake and I will be starting to create lib module to link with application. I shall open a new issue if necessary. Most of the work is about how to use open62541 for Windriver and whether to use RTPA/VIP/DKM , soon I shall share a document which can be linked to this repo or the website which might be useful for **open62541 on Windriver**.

Excited to start the measurements :) Thank you Julius and team.

closed 2019-06-18 by Pro

kingalexsylvester, 2019-08-05:
We were able to conclude that the same embedded target showed 19% better performance on VxWorks than Linux.