open62541/open62541 #2822: Generate server with the single-file distribution open62541.h/.c

cbomfim, 2019-06-17:
## Description

**# Introduction and ambient**
I am new user here, with some knowledge of C and C++ language. I tried to generate the server in two OS. First one was Ubuntu 18.04 and everything run smooth.

When I tried to generate the server from the Readme.md file in Windows 10 I run into a lot of problems. Some I could solve but I need help to go ahead..  

Downloaded zip file from Git-Revision: v0.3.0 and according tutorial, Latest commit 534ca8b 3 days ago

OS: Windows 10 Pro (Language Portuguese)

### I used CMake to have files. CMake options:

Commandline options:
-DCLANG_FORMAT_EXE:FILEPATH="CLANG_FORMAT_EXE-NOTFOUND" -DUA_BUILD_EXAMPLES:BOOL="0" -DCMAKE_CODEBLOCKS_EXECUTABLE:FILEPATH="C:/Program Files (x86)/CodeBlocks/codeblocks.exe" -DUA_ENABLE_AMALGAMATION:BOOL="1" 

Cache file:
CLANG_FORMAT_EXE:FILEPATH=CLANG_FORMAT_EXE-NOTFOUND
UA_BUILD_EXAMPLES:BOOL=0
CMAKE_CODEBLOCKS_EXECUTABLE:FILEPATH=C:/Program Files (x86)/CodeBlocks/codeblocks.exe
UA_ENABLE_AMALGAMATION:BOOL=1

## Background Information / Reproduction Steps

After that I opened project in CodeBlocks and got files open62541.c and open62541.h. After that I copied the open62541 directory of include files into include directory of MinGW.

**# Next step I run the command :**

gcc -std=c99 -DUA_ARCHITECTURE_WIN32 -lws2_32 server.c open62541.c -o server

Version of GCC: gcc (tdm-1) 4.9.2

**# I got several error messages:**

In file included from C:/Program Files (x86)/CodeBlocks/MinGW/include/open62541/server.h:18:0,
                 from server.c:2:
C:/Program Files (x86)/CodeBlocks/MinGW/include/open62541/types_generated.h:8:23: fatal error: open62541.h: No such file or directory
 #include "open62541.h"
                       ^
compilation terminated.
open62541.c: In function 'UA_NodeId_toString':
open62541.c:18021:9: warning: implicit declaration of function '_snprintf_s' [-Wimplicit-function-declaration]
         snprintfLen = UA_snprintf(nsStr, 10, "ns=%d;", nodeId->namespaceIndex);
         ^
In file included from open62541.c:28:0:
open62541.h:298:66: error: '_TRUNCATE' undeclared (first use in this function)
 #define UA_snprintf(source, size, ...) _snprintf_s(source, size, _TRUNCATE, __VA_ARGS__)
                                                                  ^
open62541.c:18021:23: note: in expansion of macro 'UA_snprintf'
         snprintfLen = UA_snprintf(nsStr, 10, "ns=%d;", nodeId->namespaceIndex);
                       ^
open62541.h:298:66: note: each undeclared identifier is reported only once for each function it appears in
 #define UA_snprintf(source, size, ...) _snprintf_s(source, size, _TRUNCATE, __VA_ARGS__)
                                                                  ^
open62541.c:18021:23: note: in expansion of macro 'UA_snprintf'
         snprintfLen = UA_snprintf(nsStr, 10, "ns=%d;", nodeId->namespaceIndex);
                       ^
open62541.c: In function 'UA_DateTime_localTimeUtcOffset':
open62541.c:53337:5: warning: implicit declaration of function 'gmtime_s' [-Wimplicit-function-declaration]
     gmtime_s(&ptm, &rawtime);
     ^
In file included from open62541.c:28:0:
open62541.c: In function 'addServerSocket':
open62541.c:53673:45: error: 'IPV6_V6ONLY' undeclared (first use in this function)
        UA_setsockopt(newsock, IPPROTO_IPV6, IPV6_V6ONLY,
                                             ^
open62541.h:269:89: note: in definition of macro 'UA_setsockopt'
 #define UA_setsockopt(sockfd, level, optname, optval, optlen) setsockopt(sockfd, level, optname, (const char*) (optval), optlen)
                                                                                         ^
In file included from open62541.c:28:0:
open62541.c: In function 'ServerNetworkLayerTCP_start':
open62541.h:298:66: error: '_TRUNCATE' undeclared (first use in this function)
 #define UA_snprintf(source, size, ...) _snprintf_s(source, size, _TRUNCATE, __VA_ARGS__)
                                                                  ^
open62541.c:53736:5: note: in expansion of macro 'UA_snprintf'
     UA_snprintf(portno, 6, "%d", layer->port);
     ^
open62541.c: In function 'UA_ClientConnectionTCP_init':
open62541.h:298:66: error: '_TRUNCATE' undeclared (first use in this function)
 #define UA_snprintf(source, size, ...) _snprintf_s(source, size, _TRUNCATE, __VA_ARGS__)
                                                                  ^
open62541.c:54195:5: note: in expansion of macro 'UA_snprintf'
     UA_snprintf(portStr, 6, "%d", port);
     ^
open62541.c: In function 'UA_ClientConnectionTCP':
open62541.h:298:66: error: '_TRUNCATE' undeclared (first use in this function)
 #define UA_snprintf(source, size, ...) _snprintf_s(source, size, _TRUNCATE, __VA_ARGS__)
                                                                  ^
open62541.c:54253:5: note: in expansion of macro 'UA_snprintf'
     UA_snprintf(portStr, 6, "%d", port);
     ^
### Another try
After some attempts I tried use the single file downloaded from Git however I run into the same problems.

Can someone help me with this?

I appreciate any help,

Carlos H M Bomfim

 - [x ] open62541 Version (release number or git tag): 534ca8b
 - [ x] Other OPC UA SDKs used (client or server): server
 - [ x] Operating system: Windows 10 Pro (Language: Portuguese)
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2019-06-17:
Add the directly where open62541.h is located as an include directory.

`gcc -I<path/to/open62541.h>`

cbomfim, 2019-06-17:
### I did that:

gcc -std=c99 -DUA_ARCHITECTURE_WIN32 -lws2_32 -I D:\Carlos\winc\opc_foundation\open62541UA\open62541UA_amal\build server.c open62541.c -o server

### And got:

                 from C:/Program Files (x86)/CodeBlocks/MinGW/include/open62541/server.h:18,
                 from server.c:2:
D:\Carlos\winc\opc_foundation\open62541UA\open62541UA_amal\build/open62541.h:18521:1: note: previous definition of 'UA_DataTypeAttributes_copy' was here
 UA_DataTypeAttributes_copy(const UA_DataTypeAttributes *src, UA_DataTypeAttributes *dst) {
 ^
In file included from C:/Program Files (x86)/CodeBlocks/MinGW/include/open62541/server.h:19:0,
                 from server.c:2:
C:/Program Files (x86)/CodeBlocks/MinGW/include/open62541/types_generated_handling.h:2568:1: error: conflicting types for 'UA_DataTypeAttributes_deleteMembers'
 UA_DataTypeAttributes_deleteMembers(UA_DataTypeAttributes *p) {
 ^
In file included from C:/Program Files (x86)/CodeBlocks/MinGW/include/open62541/types_generated.h:8:0,
                 from C:/Program Files (x86)/CodeBlocks/MinGW/include/open62541/server.h:18,
                 from server.c:2:
D:\Carlos\winc\opc_foundation\open62541UA\open62541UA_amal\build/open62541.h:18526:1: note: previous definition of 'UA_DataTypeAttributes_deleteMembers' was here
 UA_DataTypeAttributes_deleteMembers(UA_DataTypeAttributes *p) {
 ^
In file included from C:/Program Files (x86)/CodeBlocks/MinGW/include/open62541/server.h:19:0,
                 from server.c:2:
C:/Program Files (x86)/CodeBlocks/MinGW/include/open62541/types_generated_handling.h:2573:1: error: conflicting types for 'UA_DataTypeAttributes_clear'
 UA_DataTypeAttributes_clear(UA_DataTypeAttributes *p) {
 ^
In file included from C:/Program Files (x86)/CodeBlocks/MinGW/include/open62541/types_generated.h:8:0,
                 from C:/Program Files (x86)/CodeBlocks/MinGW/include/open62541/server.h:18,
                 from server.c:2:
D:\Carlos\winc\opc_foundation\open62541UA\open62541UA_amal\build/open62541.h:18531:1: note: previous definition of 'UA_DataTypeAttributes_clear' was here
 UA_DataTypeAttributes_clear(UA_DataTypeAttributes *p) {
 ^
.... and a lot of line like these above.
   
**I think I commited some mistake when when I built the system. **
Previously I generate the installation with default settings plus Examples. It work fine and I got every program which I tested. Only when I tried the single-file option I run into this problem. I attempet to work without CMake.

I have included code here to be possible to verify what I am trying to compile. They are the same we can get on line.

[build.zip](https://github.com/open62541/open62541/files/3298311/build.zip)

Thank you for this try,

Carlos H M Bomfim

jpfr, 2019-06-18:
If you use the amalgamation, you should **only** use `open62541.h`.
The server.h is including `types_generated_handling.h`.

So you see duplicate definitions.

cbomfim, 2019-06-18:
**Now duplicate has gone. Thank you for this. Helped a lot, so far.**
 
**However got new problems.** 
The _TRUNCATE definition should be at stdlib.h but, I look after that and it is not there. I will make attempt to change MinGW installation trying to solve it. I have found this when looking for correction:

https://sourceforge.net/p/mingw/bugs/2274/

This page states that MinGW don't have this. I am using MinGW64, as recommendation I read in the Opn62541 site.

_Another error I got was:_

                 open62541.c:53673:45: error: 'IPV6_V6ONLY' undeclared (first use in this function)

I don't have looking for this error yet. Still working on it.

Thanks once more,

Carlos H M Bomfim

Stasik0, 2019-06-25:
does this one and following items help? https://github.com/open62541/open62541/issues/1580#issuecomment-363091634

cbomfim, 2019-07-01:
To update this issue. 

I was able to generate Client and Server using Visual Studio 2019 Community Edition. Only thing I needed was to include the ws2_32.lib to the project. I had problem with port 4840 because another application allocated  this port. Tried use 

UA_ServerConfig *config = UA_ServerConfig_new_minimal(48880, NULL);

but received error on UA_ServerConfig_new_minimal. At the moment I changed directly in open62541.h. Not a good solution, just to move on now.

Trying to use GCC also, from Mingw64. However when I invoked gcc I got error on ws2_32.lib which says lib not found. Trying to solve that yet. I installed Mingw64 just unziping x86_64-8.1.0-release-win32-seh-rt_v6-rev0.7z into Codeblocks/Mingw directory. And set the  Codeblocks/Mingw/bin in the PATH variable.

If someone has a clue to help me I appreciate it.

Carlos Henrique de Morais Bomfim

Pro, 2019-07-23:
This is very likely a duplicate of #2877 and #1734.

Try to use the mingw64 package, that should work. Mingw32 is missing some features
https://mingw-w64.org/doku.php

Or msys2
https://www.msys2.org/

Please reopen if necessary.

closed 2019-07-23 by Pro