open62541/open62541 #2823: Intel Compiler 18.1 crash when compiling amalgamation header/source under Linux

ebissonn, 2019-06-17:
## Description
1>": : error : ** The compiler has encountered an unexpected problem.
1>  ** Segmentation violation signal raised. **
1>  Access violation or stack overflow. Please contact Intel Support for assistance.
## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=FULL -DUA_ENABLE_AMALGAMATION
```

## Checklist

Please provide the following information:

 - [0.0.4 ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ Fedora 27] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

ebissonn, 2019-06-17:
We had to rewrote the namespace_0 initialization code to get it compiling under Intel 18.1
See out PR
https://github.com/open62541/open62541/pull/2824

teccheck, 2025-03-13:
It seems there hasn't been any progress on the PR and this issue, so I'll close it. Feel free to reopen this issue and create a new PR if it's still relevant