open62541/open62541 #2826: Seg. fault when there are 0 users, UA_ServerConfig_new_basic256sha256 configuration method used

Nik654, 2019-06-18:
## Description
open62541 0.3 release version is used. Program received signal SIGSEGV, Segmentation fault when we try to create server without users, UA_ServerConfig_new_basic256sha256 configuration method is used.
## Background Information / Reproduction Steps
We modified example from “open62541-0.3.0/examples/encryption/server_basic128rsa15.c” to use configuration method “UA_ServerConfig_new_basic256sha256” instead “UA_ServerConfig_new_basic128rsa15” with the same parameters and additionally added only one line below to disable default users:
config->accessControl = UA_AccessControl_default(true, 0, NULL);

Without line above everything works and we are able to connect with default username/password created with “UA_ServerConfig_new_basic256sha256” (user1/password, user2/password1 – plugins/ua_config_default.c line 189), but when we pass for number of users 0 and UA_UsernamePasswordLogin NULL we are getting segmentation fault. We also tried with “UA_ServerConfig_new_basic128rsa15”, the same approach and it works, server rejected connection without segmentation fault because there is no users. It seems that problem is only reproducible with “UA_ServerConfig_new_basic256sha256”.

Here is the part of backtrace:
#0  0x000055555555fccd in UA_String_equal (s1=0x5555558419a0, s2=0x20) at /home/user/Desktop/open62541/open62541-0.3.0/src/ua_types.c:111
        is = 0
#1  0x00005555555a43e4 in activateSession_default (server=0x5555557f84a0, ac=0x5555557f8718, sessionId=0x55555582b1c8, userIdentityToken=0x7fffffffd270, sessionContext=0x55555582b1c0)
    at /home/user/Desktop/open62541/open62541-0.3.0/plugins/ua_accesscontrol_default.c:87
        i = 0
        userToken = 0x555555841990
        match = false
        context = 0x5555558124e0
#2  0x00005555555cf5fb in Service_ActivateSession (server=0x5555557f84a0, channel=0x555555840f10, session=0x55555582b100, request=0x7fffffffd1c0, response=0x7fffffffd100)
    at /home/user/Desktop/open62541/open62541-0.3.0/src/server/ua_services_session.c:279
No locals.
#3  0x00005555555cc6ca in processMSG (server=0x5555557f84a0, channel=0x555555840f10, requestId=3, msg=0x7fffffffd670) at /home/user/Desktop/open62541/open62541-0.3.0/src/server/ua_server_binary.c:481

Log:
./server_basic128rsa15 cert2048.der key2048.key uaexpert.der
[2019-06-18 09:18:43.844 (UTC+0200)] info/network       TCP network layer listening on opc.tcp://user-VirtualBox:4840/
[2019-06-18 09:18:47.255 (UTC+0200)] info/network       Connection 5 | New connection over TCP from 127.0.0.1
[2019-06-18 09:18:47.256 (UTC+0200)] info/channel        Creating a new SecureChannel
[2019-06-18 09:18:47.256 (UTC+0200)] warn/securitypolicy          No PKI plugin set. Accepting all certificates
[2019-06-18 09:18:47.256 (UTC+0200)] info/channel        Connection 5 | SecureChannel 1 | Opened SecureChannel
[2019-06-18 09:18:47.257 (UTC+0200)] info/channel        Connection 5 | SecureChannel 1 | CloseSecureChannel
[2019-06-18 09:18:47.257 (UTC+0200)] info/network       Connection 5 | Closed
[2019-06-18 09:18:47.271 (UTC+0200)] info/network       Connection 5 | New connection over TCP from 127.0.0.1
[2019-06-18 09:18:47.271 (UTC+0200)] info/channel        Creating a new SecureChannel
[2019-06-18 09:18:47.271 (UTC+0200)] warn/securitypolicy          No PKI plugin set. Accepting all certificates
[2019-06-18 09:18:47.271 (UTC+0200)] info/channel        Connection 5 | SecureChannel 2 | Opened SecureChannel
Segmentation fault (core dumped)

Used CMake options:

```bash
cmake -DUA_ENABLE_ENCRYPTION=ON -DUA_DEBUG=ON -DUA_BUILD_EXAMPLES=ON ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): release version 0.3
 - [x] Other OPC UA SDKs used (client or server): UA Expert 
 - [x] Operating system: Linux
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached: Small log added in issue description
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

schroeder-, 2019-06-18:
It's a bug in Version 0.3, it's already fixed in master. If the user list is empty, part of context in the accesscontrol is uninitialized.

jpfr, 2019-06-18:
Okay. We will make a fix to 0.3 as people still rely on it.

jpfr, 2019-06-18:
Hmm. I can't reproduce the problem in my box.

```c
     /* Access Control. Anonymous Login only. */
    conf->accessControl = UA_AccessControl_default(true, 0, NULL); //usernamePasswordsSize, usernamePasswords);
```
Also, the segfault in 
```c
        for(size_t i = 0; i < context->usernamePasswordLoginSize; i++) {
            if(UA_String_equal(&userToken->userName, &context->usernamePasswordLogin[i].username) &&
               UA_String_equal(&userToken->password, &context->usernamePasswordLogin[i].password)) {
                match = true;
                break;
            }
        }
```
Should not occur because `context->usernamePasswordLoginSize == 0`.

Which client do you use?

schroeder-, 2019-06-18:
` AccessControlContext *context = (AccessControlContext*)
        UA_malloc(sizeof(AccessControlContext));`
Is not memset and so is context->usernamePasswordLogin and context->usernamePasswordLoginSize undefined.

jpfr, 2019-06-18:
```c
    UA_ServerConfig *conf = (UA_ServerConfig *)UA_malloc(sizeof(UA_ServerConfig));
    if(!conf)
        return NULL;

    /* Zero out.. All members have a valid initial value */
    memset(conf, 0, sizeof(UA_ServerConfig));
```

I don't see a code-path where the config is not initialized immediately.

Did you introduce major changes to the config generation?

schroeder-, 2019-06-18:
In plugins/ua_accesscontrol_default.c in 0.3 accescontrol.context gets initialized.
Line 190
` AccessControlContext *context = (AccessControlContext*)
        UA_malloc(sizeof(AccessControlContext));`

context.usernamePasswordLoginSize and context.usernamePasswordLoginSize  are only set when usernamePasswordLoginSize > 0, otherwise its undefined.

jpfr, 2019-06-18:
You are right. I was only looking in ua_config_default.c.
The fix is in #2827.

commit 4e2c8285 (2019-06-18): Correctly initialize AccessControlContext (fix #2826)
closed 2019-06-18 by jpfr