open62541/open62541 #2828: Do client instances have shared resources?

Nik654, 2019-06-18:
## Description
We are using open62541 SDK version 0.3.
Multiple client instances (created with UA_Client_new(UA_ClientConfig config)) are created with different UA_ClientConfig configurations. Do these client instances have some shared resources (using global pointers and variables) inside open62541 SDK?
## Background Information / Reproduction Steps
Question is asked because of following use-case.
Since we know that open62541 is not thread-safe we definitely need to add mutex per client instance. 
When we create multiple client instances, if these instances share resources inside open62541 SDK than we need to add one mutex to protect access to all created client instances.

Used CMake options:

```bash
No options: cmake ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): release version 0.3
 - [x] Other OPC UA SDKs used (client or server): UA Expert
 - [x] Operating system: Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2019-06-18:
We are working on adding mutexes / critical sections internally.
A few more months...

Until then, all shared memory is constant.
For example datatype definitions.

So you can run clients independently.

Nik654, 2019-06-18:
OK, thank you for quick answer.
I am closing the issue.

closed 2019-06-18 by Nik654