open62541/open62541 #2829: Access control over endpoints

SebsSa, 2019-06-18:
Hello ,
I'm gonna need a little help getting started.

My server has different endpoints.
Depending on the selected endpoint, the client should have different access rights.

As I understood it, I can only change the access rights for the "whole" server via access control.

Could someone give me a hint as to whether it would be possible?

Thank you in advance

schroeder-, 2019-06-18:
You can write your own accesscontrol see plugins/ua_accesscontrol_default.c. In the activateSession callback you get the endpoint data und use it to control the access of the session.

SebsSa, 2019-06-18:
Great, thanks! I'll give it a try right now.

SebsSa, 2019-06-24:
Unfortunately my abilities are not enough to understand the whole "code". Not to mention to find an answer to my question. 

I have rummaged a bit and tried a lot but unfortunately I didn't find a solution...

Is there an example or a simple solution?
Something like :
_..endpoint[0].setAccessLevel.._
```
   for(int i = 0; i < config->endpointsSize; i++) 
    {
    if(config->endpoints[i].endpointDescription.securityMode == UA_MESSAGESECURITYMODE_NONE) 
        {   
           config->endpoints[i].setAccessLevel = ... (0,1,2)(readonly,writeRead,...)
        }
    }
```
**!This is no real code!**

It would be great if someone could recommend a helpful link to me. Thank you once in advance

jpfr, 2019-06-24:
Here is an example where the standard AccessControl policy (allow all) is overriden.

https://github.com/open62541/open62541/blob/master/examples/access_control/server_access_control.c

When the session is created, the AccessControl plugin inspects the login information and can attach custom data to the session.

This data is then evaluated at runtime.
When a service (like Read) is called on a node in the information model, the AccessControl plugin is queried if the user has sufficient rights.

Feras992, 2019-08-09:
@jpfr hello,
is it possible to know which user is connected, so i give for each on of them different permissions?
for example in the example above .. can i know if peter is connected so i give him just the permissions to just read the Vairables but not to write them and not to call methods
and when paula is connected she allowed to read and write the Variables and call the methods ?

mlang-de, 2019-09-27:
@Feras992 this you can also solve with the custom data/sessionContext from the AccessControl plugin. During ActivateSession you must store the relevant/session-data like username or permission groups in you own defined sessionContext.

closed 2021-09-14 by SebsSa