open62541/open62541 #2831: Nodes that have a value callback do not work if immutable nodes is enabled

cvals, 2019-06-19:
## Description
If UA_ENABLE_IMMUTABLES_NODES=ON then reading nodes that have a value callback results in: `ua_nodestore_default.c:166: UA_Nodestore_releaseNode: Assertion entry->refCount > 0 failed.`

## Background Information / Reproduction Steps
Reading "Current time - value callback" in the server generated by the following, results in the error, if UA_ENABLE_IMMUTABLE_NODES=ON.
  
```c
#include <open62541/plugin/log_stdout.h>
#include <open62541/server.h>
#include <open62541/server_config_default.h>

#include <signal.h>
#include <stdlib.h>

static volatile UA_Boolean running = true;
static void stopHandler(int sig) {
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "received ctrl-c");
    running = false;
}
static void
updateCurrentTime(UA_Server *server) {
    UA_DateTime now = UA_DateTime_now();
    UA_Variant value;
    UA_Variant_setScalar(&value, &now, &UA_TYPES[UA_TYPES_DATETIME]);
    UA_NodeId currentNodeId = UA_NODEID_STRING(1, "current-time-value-callback");
    UA_Server_writeValue(server, currentNodeId, value);
}

static void
beforeReadTime(UA_Server *server,
               const UA_NodeId *sessionId, void *sessionContext,
               const UA_NodeId *nodeid, void *nodeContext,
               const UA_NumericRange *range, const UA_DataValue *data) {
   updateCurrentTime(server);
}

static void
afterWriteTime(UA_Server *server,
               const UA_NodeId *sessionId, void *sessionContext,
               const UA_NodeId *nodeId, void *nodeContext,
               const UA_NumericRange *range, const UA_DataValue *data) {
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND,
                "The variable was updated");
}

static void
addValueCallbackToCurrentTimeVariable(UA_Server *server) {
    UA_NodeId currentNodeId = UA_NODEID_STRING(1, "current-time-value-callback");
    UA_ValueCallback callback ;
    callback.onRead = beforeReadTime;
    callback.onWrite = afterWriteTime;
    UA_Server_setVariableNode_valueCallback(server, currentNodeId, callback);
}

static void
addCurrentTimeVariable(UA_Server *server) {
    UA_DateTime now = 0;
    UA_VariableAttributes attr = UA_VariableAttributes_default;
    attr.displayName = UA_LOCALIZEDTEXT("en-US", "Current time - value callback");
    attr.accessLevel = UA_ACCESSLEVELMASK_READ | UA_ACCESSLEVELMASK_WRITE;
    UA_Variant_setScalar(&attr.value, &now, &UA_TYPES[UA_TYPES_DATETIME]);

    UA_NodeId currentNodeId = UA_NODEID_STRING(1, "current-time-value-callback");
    UA_QualifiedName currentName = UA_QUALIFIEDNAME(1, "current-time-value-callback");
    UA_NodeId parentNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);
    UA_NodeId parentReferenceNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES);
    UA_NodeId variableTypeNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE);
    UA_Server_addVariableNode(server, currentNodeId, parentNodeId,
                              parentReferenceNodeId, currentName,
                              variableTypeNodeId, attr, NULL, NULL);
}

int main(void) {
    signal(SIGINT, stopHandler);
    signal(SIGTERM, stopHandler);

    UA_Server *server = UA_Server_new();
    UA_ServerConfig_setDefault(UA_Server_getConfig(server));
    addCurrentTimeVariable(server);
    addValueCallbackToCurrentTimeVariable(server);
    UA_StatusCode retval = UA_Server_run(server, &running);

    UA_Server_delete(server);
    return retval == UA_STATUSCODE_GOOD ? EXIT_SUCCESS : EXIT_FAILURE;
}
```

The error is caused by the call to UA_Nodestore_releaseNode here https://github.com/open62541/open62541/blob/3b14df892c538edc674afd09325b7c6f1ede51f6/src/server/ua_services_attribute.c#L392

I think the problem is because the node is already released here https://github.com/open62541/open62541/blob/3b14df892c538edc674afd09325b7c6f1ede51f6/src/server/ua_services_attribute.c#L129

Shouldn't this one be released instead in readValueAttributeFromNode? https://github.com/open62541/open62541/blob/3b14df892c538edc674afd09325b7c6f1ede51f6/src/server/ua_services_attribute.c#L128

I think changing readValueAttributeFromNode to something like this would fix this? 

```c
static UA_StatusCode
readValueAttributeFromNode(UA_Server *server, UA_Session *session,
                           const UA_VariableNode *vn, UA_DataValue *v,
                           UA_NumericRange *rangeptr) {
    if(vn->value.data.callback.onRead) {
        vn->value.data.callback.onRead(server, &session->sessionId,
                                       session->sessionHandle, &vn->nodeId,
                                       vn->context, rangeptr, &vn->value.data.value);
        /* Reopen the node to see the changes from onRead */
        vn = (const UA_VariableNode*)UA_Nodestore_getNode(server->nsCtx, &vn->nodeId);
    }
    if(rangeptr)
        return UA_Variant_copyRange(&vn->value.data.value.value, &v->value, *rangeptr);
    *v = vn->value.data.value;
    v->value.storageType = UA_VARIANT_DATA_NODELETE;
    if(vn->value.data.callback.onRead)
        UA_Nodestore_releaseNode(server->nsCtx, (const UA_Node *)vn);
    return UA_STATUSCODE_GOOD;
}
```

Used CMake options:

```bash
cmake -UA_ENABLE_IMMUTABLE_NODES=ON ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2019-06-19:
Quick questions.
Do you access the server from parallel threads?

cvals, 2019-06-19:
> Quick questions.
> Do you access the server from parallel threads?
 
Yes

jpfr, 2019-06-19:
Then the problem you report makes sense.
Multithreading-support is incomplete. (And we give a warning if that is activated, imho.)

Your best option is to protect the server with a lock/mutex.
So that only one process can access the server at a time.
C++ example code for that was posted in #2783.

We are actively working on this.
But it will take a few more months to land.

Closing as a duplicate of #2783.

closed 2019-06-19 by jpfr

cvals, 2019-06-19:
> Then the problem you report makes sense.
> Multithreading-support is incomplete. (And we give a warning if that is activated, imho.)
>
> Your best option is to protect the server with a lock/mutex.
> So [...]

Ok, but you also get this problem when you run single threaded and enable immutable nodes.
But using immutable nodes without parallel threads is maybe not very useful