open62541/open62541 #2832: Build Error on MacOS

joesan, 2019-06-19:
## Description 
Error when trying to build from the sources on Mac OS Mojave

## Background Information / Reproduction Steps
Followed the guide as mentioned in the documentation

Used CMake options:
```
Joes-MacBook-Pro:build joesan$ cmake ..
-- Architectures included in amalgamation: posix
-- The selected architecture is: posix
-- Could NOT find LATEX (missing: LATEX_COMPILER)
CMake Error at CMakeLists.txt:1072 (message):
  Examples cannot be built with source amalgamation enabled

-- Configuring incomplete, errors occurred!
See also "/Users/joesan/Projects/Private/c-projects/open62541/CMakeFiles/CMakeOutput.log".
Joes-MacBook-Pro:build joesan$
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): latest from git master
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system: Mac OS Mojave
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2019-06-19:
That is not MacOS specific.

> Examples cannot be built with source amalgamation enabled

Either disable the examples or use "normal" compilation instead of the amalgamation.

joesan, 2019-06-19:
Ok that helped! Here is the next one:

```
Joes-MacBook-Pro:build joesan$ cmake ..
-- The selected architecture is: posix
-- Could NOT find LATEX (missing: LATEX_COMPILER)
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/joesan/Projects/Private/c-projects/open62541
```

Is this Okay? with the missing LATEX_COMPILER? Is this for some documentation purposes?

I was able to build the project now! We can close this issue!

closed 2019-06-19 by joesan