open62541/open62541 #2834: Replacing Malloc

JustinKieu, 2019-06-19:
Hello,
I am attempting to change the library from dynamic allocation to static allocation for a bare metal environment. I have tried reading this issue #1982 , but am unable to figure out how to replace malloc as UA_config.h does not exist. We have written our own implementation for malloc, but cannot figure out how to implement it in.
Thanks!

Please provide the following information:

 - [x] open62541 Version (release number or git tag): master branch
 - [x] Operating system: Windows

jpfr, 2019-06-19:
See here: https://github.com/open62541/open62541/blob/3b14df892c538edc674afd09325b7c6f1ede51f6/arch/posix/ua_architecture.h#L113

Or in the corresponding architecture folder.

jpfr, 2019-06-24:
UA_config.h is <open62541/config.h> and is generated by cmake from /include/open62541/config.h.in.

JustinKieu, 2019-06-24:
Thank you for the help!

closed 2019-06-24 by JustinKieu