open62541/open62541 #2835: Historize data in a circular buffer

luibass92, 2019-06-20:
Goodmorning,

I am a "Historical Data" plugin user and I see that there is no limit on the size of the buffer which historizes data.
I can set an initial size but, each time it is reached, a UA_realloc is done and the buffer is re-allocated with its size doubled.

I would like to configure the historical data in order to have something similar to a circular buffer beacause I want to be sure to not have undesired memory growths.

I managed how to do it with the Gathering, now I can register only a maximum number of nodes and if I overcome that limit an UA_STATUS_BADOUTOFMEMORY is returned.
Now I am trying to do the same with the Backend part but it seems way more complicated.

Someone could please help me or give me some tips?

Thanks,
Luigi

jpfr, 2019-06-24:
@peter-rustler-basyskom 
@basyskom-meerkoetter 

Can you have a look at this?

gcn1996, 2019-12-12:
I have the same question,it will take too much memory and break the program

luibass92, 2020-05-29:
Hi everyone,
after several months I have implemented by myself a solution to have a "circular history".

The idea behind it is to have a fixed number of nodes to historize, that means it is mandatory to choose a maximum number of nodes at compile time (and that number will never be exceeded) and for each of that nodes a circular array of values will be saved and, each time a client performs a UA_HistoryReadRequest on a certain node, the current circular buffer (full of values) will be sent by the server.

To do so I started from the original history functions and I did some modifications to obtain the desired result. This was the faster solution for me but it means that most of the code it is very similar to the original implementation. Sorry for that but I did not found a more clever and fast way to implement the algorithms.

The following images describe the differences between the original history and the "circular" one.

Original implementation of OPC UA Historical Data Access: nodes and
corresponding values can grow indefinitely:

[image]

Modified implementation of OPC UA Historical Data Access: if max
number of nodes is reached (in the image is 3), registration of other nodes is not allowed. If max
number of values is reached (in the image is 4), subsequent values are saved overwriting oldest ones
following the rules of circular buffers

[image]

Moreover it is useful to know that I have added a field to the `UA_NodeIdStoreContextItem_backend_memory` struct in order to make things easier to implement.

To try this implementation all you need to do is to set:

```
serverSetHistoryData = &serverSetHistoryData_backend_memory_Circular;
getHistoryData = &getHistoryData_service_Circular;
registerNodeId = &registerNodeId_gathering_Circular;
```
While the History Database part (interface) will remain untouched.

Find attached a zip file with the code, your opinions would be very appreciated ^_^

[circular_history_open62541.zip](https://github.com/open62541/open62541/files/4702120/circular_history_open62541.zip)

@jpfr 
@peter-rustler-basyskom 
@basyskom-meerkoetter

luibass92, 2021-10-21:
> Hi I am trying to implement a circular buffer. But I cannot find the functions UA_DataValueMemoryStoreItem_deleteMembers and UA_NodeIdStoreContextItem_deleteMembers. Did you write them yourself? They are not in the ZIP file. Or is UA_DataValueMemoryStoreItem_deleteMembers =UA_DataValueMemoryStoreItem_clear and UA_NodeIdStoreContextItem_deleteMembers =UA_NodeIdStoreContextItem_clear? Thanks a lot!

Hi @Max15421,
first of all thanks for re-opening this topic, it seems that not so many people need to avoid memory leaks in their code 😄 
As far as I remember I have been using the default functions UA_DataValueMemoryStoreItem_deleteMembers and UA_NodeIdStoreContextItem_deleteMembers .
This is a general rule, if I did not implement some of the functions it means that or I am not calling them at all (in the interface they are set to NULL), or I am using the default ones.

Have you managed to compile the newest version of open62541 with these modifications?

luibass92, 2021-11-01:
I finally decided to create a pull request after adapting the code with the latest version of the library.
You can find the pull request [here](https://github.com/open62541/open62541/pull/4740).
Thanks for your support :)

luibass92, 2021-11-17:
Merged on master https://github.com/open62541/open62541/pull/4740.

Closing this issue.

closed 2021-11-17 by luibass92