open62541/open62541 #2839: Subscriber: Using multiple ReaderGroups causes PubSubChannel->regist to fail

Fischer-Thomas, 2019-06-24:
There is an issue regarding the subscriber implementation. After I tried to use multiple DataSetReader in a single ReaderGroup (#2817) and ran into some problems I tried to use multiple ReaderGroups with single DataSetReaders. 

However this doesn't work and gives me the following error log: 

> [2019-06-24 15:15:53.941 (UTC+0200)] error/server	PubSub Connection regist failed.
>[2019-06-24 15:15:53.941 (UTC+0200)] warn/server	register channel failed: 0x80020000!

It seems that regist fail due to an error in the bind function.
The regist function is called inside the "UA_ReaderGroup_addSubscribeCallback" function which is performed everytime when you add a ReaderGroup using the "UA_Server_addReaderGroup" API function. 

Thus, when having multiple ReaderGroups, there are multiple binds on the same socket which causes the error. (The first regist returns with no errors but every other after that..)

https://github.com/open62541/open62541/blob/2cfff7aa4f513e661a718de5b561290b79c1b207/src/pubsub/ua_pubsub.c#L209-L243

https://github.com/open62541/open62541/blob/2cfff7aa4f513e661a718de5b561290b79c1b207/src/pubsub/ua_pubsub.c#L2406-L2411

jpfr, 2019-06-24:
@opcua-tsn-team-kalycito 
Please have a look.
Also, cover this with a unit test to prevent regressions.

Thanks.

opcua-tsn-team-kalycito, 2019-08-21:
Hi @Fischer-Thomas,
We have addressed this issue in PR #2944.
The same has been merged to the master.

Regards
opcua-tsn-team-kalycito

closed 2021-04-07 by andreasebner