open62541/open62541 #2842: connect UA server

xiaotian711, 2019-06-25:
connect  UA server by the way of username and password,,I am a student,I can't modify my C++ program of connection, please do me a favor! thanks!

## Description connect  UA server by the way of username and password,,I am a student,I can't modify my C++ program of connection, please do me a favor! thanks!

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): the lateset
 - [ ] Other OPC UA SDKs used (client or server): client
 - [ ] Operating system:win10
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue connect server:
#include <stdio.h>
#include <iostream>
#include "open62541.h"
#pragma comment(lib,"ws2_32.lib")    
int main()
{
	UA_Client *client = UA_Client_new(UA_ClientConfig_default);

	     
	UA_EndpointDescription* endpointArray = NULL;
	size_t endpointArraySize = 0;
	UA_StatusCode retval = UA_Client_getEndpoints(client, "opc.tcp://localhost:4840",
		&endpointArraySize, &endpointArray);
	if (retval != UA_STATUSCODE_GOOD) {
		UA_Array_delete(endpointArray, endpointArraySize, &UA_TYPES[UA_TYPES_ENDPOINTDESCRIPTION]);
		UA_Client_delete(client);
		return (int)retval;
	}
	printf("%i endpoints found\n", (int)endpointArraySize);
	for (size_t i = 0; i < endpointArraySize; i++) {
		printf("URL of endpoint %i is %.*s\n", (int)i,
			(int)endpointArray[i].endpointUrl.length,
			endpointArray[i].endpointUrl.data);
	}
	UA_Array_delete(endpointArray, endpointArraySize, &UA_TYPES[UA_TYPES_ENDPOINTDESCRIPTION]);

	
	retval = UA_Client_connect_username(client, "opc.tcp://localhost:4840", "user1", "password");
	if (retval != UA_STATUSCODE_GOOD) {
		UA_Client_delete(client);
		return (int)retval;
	}

}

Pro, 2019-06-25:
It's not clear what you are asking. Please have a look at our examples. There you find client and server code for user/password connections.

https://github.com/open62541/open62541/tree/master/examples

Closing as invalid

closed 2019-06-25 by Pro

xiaotian711, 2019-06-26:
examples-client.c?        May I have your WeChat?  I don't have reference materials about server connection, I need your help,,        there is a numerically controlled production center that integrate the ua server, my task is write c++ code to connect ua server and read/write data,  I'm trying to use Example Client Implementation to connect, but i didn't know whether the code is true or how to judge connection successful.

xiaotian711, 2019-06-26:
> It's not clear what you are asking. Please have a look at our examples. There you find client and server code for user/password connections.
> 
> https://github.com/open62541/open62541/tree/master/examples
> 
> Closing as invalid

> It's not clear what you are asking. Please have a look at our examples. There you find client and server code for user/password connections.
> 
> https://github.com/open62541/open62541/tree/master/examples
> 
> Closing as invalid

examples-client.c? May I have your WeChat? I don't have reference materials about server connection, I need your help,, there is a numerically controlled production center that integrate the ua server, my task is write c++ code to connect ua server and read/write data, I'm trying to use Example Client Implementation to connect, but i didn't know whether the code is true or how to judge connection successful.