open62541/open62541 #2843: Running the example Server on my Mac

joesan, 2019-06-26:
## Description 
Unable to run the example server as mentioned in the documentation here: https://open62541.org/doc/0.3/tutorial_server_firststeps.html

## Background Information / Reproduction Steps
I was trying to get the server up and running on my Mac. I tried to do this as a Docker container. Here is the `[Dockerfile](https://github.com/joesan/opc-client-stream/blob/master/Dockerfile)` that I'm using!

When I build my Docker image, I get the following error:

```
Step 11/12 : RUN gcc -std=c99 open62541.c myServer.c -lws2_32 -o myServer.exe
 ---> Running in f95b293d0fd0
/usr/bin/ld: cannot find -lws2_32
collect2: error: ld returned 1 exit status
```

Any ideas on how I can load this ws2_32 library?

## Checklist
 - [x] open62541 Version (release number or git tag): 0.3.0
 - [x] Operating system: Mac OS Mojave

jpfr, 2019-06-26:
ws2_32 is a Windows library. Your mac does not have it.

joesan, 2019-06-26:
So to run the example server, I need to be on Windows is it? How can I run it on Ubuntu?

jpfr, 2019-06-26:
The library works in Mac, Linux, Windows and a couple more.
But you don't need include ws2_32 on Linux. It should simply work without.

Please reopen if questions remain.

closed 2019-06-26 by jpfr