open62541/open62541 #2844: [OPCUA CLIENT]: "UA_Client_getEndpoints"  return "BadSecureChannelIdInvalid"

elitezhu, 2019-06-26:
## Description
I use open6541 as OPCUA client, and connect a B&R X20 PLC which acts OPCUA server.
I refer the example of https://github.com/open62541/open62541/blob/master/examples/client.c
But when it starts to get the endpoint, the API, "UA_Client_getEndpoints"  return "BadSecureChannelIdInvalid". I also use the UAExpert to connect the server, it's OK. Besides, I also use the OPCUA lib from C# core to do the same thing, it's also OK.
So I'm wondering what's wrong.

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):master
 - [ ] Other OPC UA SDKs used (client or server):  client
 - [ ] Operating system: linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

opcua-tsn-team-kalycito, 2019-06-26:
@elitezhu in both cases, open62541 client as well as UAExpert client, can you share the Wireshark trace?

Regards,
@opcua-tsn-team-kalycito

elitezhu, 2019-06-27:
@opcua-tsn-team-kalycito In the customer's plant, there's 3 same type PLCs. And two is existed, another is new added. The open62541 can't get the endpoint from the new one, but other two is OK. For the new one, UaExpert and C# core OPCUA lib can get the endpoint well. It's very strange. Actually the PLC is working in the customer's plant, so it's hard to get the Wireshark trace directly. I have asked the customer send me this PLC to test. And Could you give me some hints?

opcua-tsn-team-kalycito, 2019-06-27:
@elitezhu We understand your constraint in sharing Wireshark traces. While we wait to receive your traces, could you share your open62541 client log with log configuration set at UA_LOGLEVEL=100 for both cases, i.e. the old PLC (working) and new PLC (not-working).

Regards,
@opcua-tsn-team-kalycito

elitezhu, 2019-07-01:
@opcua-tsn-team-kalycito , I have captured the raw telegram via wireshark, please have a check.
[B&R_PLC_OPCUA_RAW_TELEGRAM.zip](https://github.com/open62541/open62541/files/3344817/B.R_PLC_OPCUA_RAW_TELEGRAM.zip)

elitezhu, 2019-07-02:
@opcua-tsn-team-kalycito And this is the uaexpert telegram with the PLC via wirshark.
[B&R_PLC_Uaexpert_Telegram.zip](https://github.com/open62541/open62541/files/3347804/B.R_PLC_Uaexpert_Telegram.zip)

opcua-tsn-team-kalycito, 2019-07-02:
@elitezhu, Thanks for the Wirehark trace, can you also share the client logs with log configuration set at UA_LOGLEVEL=100 for open62541 client in both cases? One case where open62541 client is connecting successfully with the older PLC and the other case where open62541 client is throwing error code "BadSecureChannelIdInvalid" with the newer PLC.

Regards
@opcua-tsn-team-kalycito

mlgiraud, 2019-07-02:
The only notable difference i see is that the open62541 client sends null array instead of empty arrays.
You could try changing the arrays to empty arrays in the getendpoints request to see if this solves the problem.
I couldn't find anything in the standard that fobids null arrays, so it could be that this is a misbehavior on the devices side.

elitezhu, 2019-07-03:
@mlgiraud How can I change the arrays to empty? which function ?

elitezhu, 2019-07-04:
@mlgiraud I have update the array size to 0, and the telegram looks almost the same as the one of UaExpert, but it still fails. Very Strange.

[B&R_PLC_Open62541_Telegram_2019_07_04.zip](https://github.com/open62541/open62541/files/3358328/B.R_PLC_Open62541_Telegram_2019_07_04.zip)
[B&R_PLC_Uaexpert_Telegram_2019_07_04.zip](https://github.com/open62541/open62541/files/3358329/B.R_PLC_Uaexpert_Telegram_2019_07_04.zip)

@opcua-tsn-team-kalycito Sorry, I can't get the debug log of the old PLCs, Here is the log of the newer one. Please have a check.
[open62541_log.zip](https://github.com/open62541/open62541/files/3358336/open62541_log.zip)

korny86, 2020-07-21:
Did you find a solution? We have a similar problem with a B&R X20 PLC and BadSecureChannelIdInvalid.

MaoniinoaM, 2020-07-31:
> Did you find a solution? We have a similar problem with a B&R X20 PLC and BadSecureChannelIdInvalid.

I checked the issue with engineer from B&R, obviously, it comes the server of B&R, and they just said they would keep solving. But they didn't reply me until now. So you'd better ask some support from B&R.

teccheck, 2025-02-25:
This issue is outdated as the code has changed a lot since 2020. Feel free to reopen this issue if needed

closed 2025-02-25 by teccheck