open62541/open62541 #2845: Boolean value doesn't change to true in third party client

HamidMapna, 2019-06-26:
Hi everybody.
I have again faced an issue concerning the boolean data type in QNX platform. Couple of months ago I was grappling with this type data in PR (#1840) regarding the fact that my project has been working under QNX platform in which `sizeof(bool) = 4` rather than being equal to 1 (like Windows and Linux). Of course I fixed the issue in my own way and thanks to @jpfr suggested revising the issue in successive versions.
Two days ago I added a boolean type tag in OpcUA server and used UaExpert as my third party client to monitor the added tag. But surprisingly I couldn't change the default `false` value to `true`. Project was working correctly for all other data types, except boolean type. I again took a deep effort and dug the source code to iron out the problem. Eventually I figured out that `true` value doesn't properly casted to `UA_Byte*` in a `UA_DataValue.value.data` byte pointer variable. 
By revising the following lines in source code inside the `sampleCallbackWithValue` method I managed to fix the aforementioned issue .

`static UA_Boolean `
`sampleCallbackWithValue(UA_Server *server, UA_MonitoredItem *monitoredItem, UA_DataValue *value){`
`if((*(bool*)value->value.data) == true){`
`if((*(uint8_t*)value->value.data) == '\0'){`
`        UA_DataValue tempValue;`
`UA_DataValue_Init(&tempValue);`
`UA_Variant_setScalar(&tempValue.value, &value->value, &UA_TYPES[UA_TYPES_BOOLEAN]);`
`uint8_t *tempUint = &value->value.data[0];`
`*tempUint = 49;  //code ascii '1' and don't know why?`
`*tempUint = &value->value.data[1];`
`*tempUint = 49;  //code ascii '1'`
`*tempUint = &value->value.data[2];`
`*tempUint = *(uint8_t*) tempValue.value.data; // <--72 code ascii 'H'`
`}  }`
` UA_Boolean changed = detectValueChanged(...);`
`....}`
It appeared that code ascii of corresponding correct byte values for `true` value equal to 49,49 and 72 for the first, second and third bytes respectively as shown above.
Anyway, I coped with the issue and am not really sure if it concerns to underlying QNX platform or not. I appreciate any explanation.

Mebus, 2021-12-03:
Hi HamidMapna,

currently I am facing a very similar problem when connecting to a Beckhoff PLC which uses the UaExpert gateway under the hood. I can write all sorts of variables using `UA_Client_writeValueAttribute` except for the boolean type. After writing to the boolean variable the TwinCAT online view doesn't display "TRUE" or "FALSE", but "Value of the expression cannot be retrieved." 😒 

Mebus

jpfr, 2025-02-20:
This is a very old bug.
There is now QNX support on the master branch via #6967.
This hasn't been reported since, so I'm closing.
Please reopen if required.

closed 2025-02-20 by jpfr