open62541/open62541 #2846: Performance measurement of Client/Server and Pub/Sub on VxWorks

kingalexsylvester, 2019-06-26:
## Description
I am trying to measure the performance of open62541 Client/Server and Pub/Sub on embedded platform and I need the calculations of Min/Max/Avg to be done in the device itself due to the constraints of memory. Currently I am developing this logic. Is this already available somewhere or need to be developed?

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): Current master
 - [ ] Operating system: VxWorks

opcua-tsn-team-kalycito, 2019-06-27:
@kingalexsylvester  Performance measurement in VxWorks isn’t yet in our open source roadmap. However, we have already taken performance measurements in Linux and can help you ramp up quickly in your environment with more details over a call. You can connect with us at opcua-tsn.team@kalycito.com

kingalexsylvester, 2019-06-27:
I calculated Min/Max/Avg by using simple clock_gettime( ) and counter loopback application, it seems to work fine and I am trying to validate the values on hardware using the below logic:

I created a loop app like below:

N1(Pub)------------------->N2(Sub)
        |                                   |
(Counter++)                        |if(diffValue){toggle_Pin0}------>Oscilloscope
        |                                   |
N1(Sub)<--------------------N2(Pub)

I shall share the snippet once this is validated. @opcua-tsn-team-kalycito I will reach to you if help needed.

kingalexsylvester, 2019-07-04:
Hello All:

Here is the implementation. Please feedback.
```cpp
/*
* This implementation was done to measure the performance of the Publisher
* Subscriber implementation between two nodes on a VxWorks OS environment.
* The application logic is that, sending a counter data from NODE 1 and the
* NODE 2 receives it to compare if it is an incremented value. The same value
* is sent back to NODE 1, now NODE 1 gets the value and increments it by 1 and
* sends it back to NODE 2 where the round-trip measurement is done if it is an
* incremented value. This snippet was verfied with hardwared signal toggle.
*
* Notes: 
* Here both NODE 1 and NODE 2 will be running both Publisher and Subscriber applications
* NODE 1: Will receive the data and increment it by 1 and send to NODE 2
* NODE 2: Receives the data and checks if it is incremented and measurement is done.
*         After which the value is sent back to NODE 1
*/

//Global declerations
struct timespec t1;
unsigned long long newRoundTripTime, maxRoundTripTime, minRoundTripTime;
unsigned long long prevCounterVal = 1;

/*
* Implement the below code snippet in your subscriber function
* that is running on node 2. Since I have used my own subscriber
* implementation, I haven't shared my code sections which might confuse others.
*/
void yourSubscriberFunction()
{

    unsigned long long  counterVal; //the counter from node 1
    struct timespec t2;             //to store the current time
	
	/*
	* store the subscribed counter value to the
	* counterVal variable. Here we have done our own
	* subscriber implementation so I haven't added the
	* code sections which might confuse people.
	*/
    counterVal = *(UA_UInt64 *)dsm->data.keyFrameData.dataSetFields[fieldCount].value.data;

	/*
	* NOTE: The assumption is that round trip value
	* will not exceed more than 1 second. If your application
	* needs it then you go to include the subtraction of sec
	* and nsec seperately.
	*/
   if (prevCounterVal == (counterVal-1))
   {
   	    clock_gettime(CLOCK_REALTIME, &t2);
   	    newRoundTripTime = getTspecDiff(t1, t2);
   	    
   	    if (newRoundTripTime > maxRoundTripTime)
   	    {
   	    	/*
   	    	* View the value using live debug
   	    	* (or)
   	    	* print the value every 10 minutes if
   	    	* profiling is being done for 10M+ cycles
   	    	*/
   	    	maxRoundTripTime = newRoundTripTime;
   	    }
   	    else if (newRoundTripTime < minRoundTripTime)
   	    {
   	    	/*
   	    	* View the value using live debug
   	    	* (or)
   	    	* print the value every 10 minutes if
   	    	* profiling is being done for 10M+ cycles
   	    	*/
   	    	minRoundTripTime = newRoundTripTime;
   	    }
   }
   
   /*
   * Store the time and 
   * (or)
   * print the value every 10 minutes if
   * profiling is being done for 10M+ cycles
   */
   t1 = t2;
   prevCounterVal = counterVal;

}

/*
* This function returns the difference of two time
* values in the order of nano seconds.
* t1: old time
* t2: current time/ future to t1
*/
static unsigned long long getTspecDiff(struct timespec t1, struct timespec t2)
{
	unsigned long long diff = NSEC_TO_SEC * t2.tv_sec + t2.tv_nsec;

	diff = diff - (NSEC_TO_SEC * t1.tv_sec + t1.tv_nsec);
	
	return diff;
}
```

kusumy, 2020-03-12:
Hi @kingalexsylvester what about client/server performance measurements ?
I want to measure performance on Raspberry PI 2 as a Server, and Windows/Linux as a client.
Can you give any suggestions the right and easier method the measure performance on my hardware ?

Regards

closed 2026-10-20 by o6-nbeier