open62541/open62541 #2847: Error when running the Client example

joesan, 2019-06-27:
## Description

Error when building the client as mentioned in this example documentation: https://open62541.org/doc/0.3/tutorial_client_firststeps.html

## Background Information / Reproduction Steps

When I tried to run the client example., using the command below (as mentioned in the documentation), I get the following error:

```
Step 10/11 : RUN gcc -std=c99 open62541.c myClient.c -o myClient
 ---> Running in 94da21fb419b
myClient.c:1:10: fatal error: open62541/client_config_default.h: No such file or directory
 #include <open62541/client_config_default.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

I guess it was looking for some config file, but seems the file is not located in a folder format that it expects. In my case, I have the myClient.c and the open62541.c and open62541.h in the same folder. Is this correct? If so then why do I get this error?

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 0.3.0
 - [x] Operating system: MacOS Mojave

jpfr, 2019-06-27:
Hello @joesan ,

you can use open62541 in two ways:

- Using the feature-based headers
- Using the amalgamated header (open62541.h).
The amalgamated header combines all public .h files.

You cannot mix the two. If you use open62541.h, use it everywhere.
And not the feature-based headers.

joesan, 2019-06-27:
So what do I do in my case? Should I copy some additional files to the same location where the myClient.c is located?

Pro, 2019-06-27:
make sure that your are using the 0.3 version, and not the current master version.

To see how to build on OS X, you can also check our travis config:

https://github.com/open62541/open62541/blob/0.3/tools/travis/travis_osx_script.sh

Especially the first block should work on your system:

```
echo "Full Namespace 0 Generation" && echo -en 'travis_fold:start:script.build.ns0\\r'
mkdir -p build
cd build
cmake -DCMAKE_BUILD_TYPE=Debug -DUA_ENABLE_FULL_NS0=ON -DUA_BUILD_EXAMPLES=ON  ..
make -j
cd .. && rm -rf build
echo -en 'travis_fold:end:script.build.ns0\\r'
```

If that doesn't work, then please check out a new untouched version

joesan, 2019-06-27:
@Pro I will give that a try!

I see two different documentations on how to run the client:

https://open62541.org/doc/current/tutorial_client_firststeps.html - This fails with the config header not found error

https://open62541.org/doc/0.3/tutorial_client_firststeps.html - This one seems to work for me!

Pro, 2019-06-27:
Yep, if you are using 0.3 you also need to use the corresponding docs for 0.3

current is for the master version where the api and includes changed.

I assume the issue is fixed. Please reopen if necessary.

closed 2019-06-27 by Pro