open62541/open62541 #2848: Assertion Fail in UA_Nodestore_releaseNode while calling UA_Server_readDataType

juangburgos, 2019-06-27:
## Description

I have been having this issue for a debug assertion violation in `UA_Nodestore_releaseNode`:

```c
UA_assert(entry->refCount > 0);
```

If course it does not affect the release builds being an assertion, but I cannot run a server un debug mode for more than a couple of hours without it crashing the session.

What could be causing it? Maybe I am using the library API incorrectly?

Thanks.

## Background Information / Reproduction Steps

Call `UA_Server_readDataType` to read  every once in a while and it will eventually happen.

[image]

[image]

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): master
 - [x] Other OPC UA SDKs used (client or server): UaExporer
 - [x] Operating system: Windows

Pro, 2019-06-27:
Could it be that you are accessing `m_qUaServer->m_server` in different threads?

This assertion normally fails, if there are some race conditions, or (less likely) there is a missing `Node_release` in the internal code. Since you are using the public API which is tested via unit tests, that's less likely, but still possible.

juangburgos, 2019-06-27:
Not that I am aware, but I will double check my threaded code for race conditions. Will report back if find something. Thanks for the reply.

juangburgos, 2019-07-04:
Put some extra guards on threaded code and have not been able to reproduce again. Closing now, will reopen if comes back. Thank you!

closed 2019-07-04 by juangburgos