open62541/open62541 #2850: Remove multiple compiler flags for new architecture does not work

gbeuning, 2019-06-28:
## Description
Not possible to remove multiple compiler flags for new architecture with ua_architecture_remove_definitions().

## Background Information / Reproduction Steps
The problem seems to be in `Architectures changes` section in the general CMakeLists.txt:

```
GET_PROPERTY(ua_architecture_remove_definitions GLOBAL PROPERTY UA_ARCHITECTURE_REMOVE_DEFINITIONS)
if (NOT "${ua_architecture_remove_definitions}" STREQUAL "")
  string(REPLACE " " ";" ua_architecture_remove_definitions_list ${ua_architecture_remove_definitions})
  remove_definitions(${ua_architecture_remove_definitions_list})
endif(NOT "${ua_architecture_remove_definitions}" STREQUAL "")
```

$(ua_architecture_remove_definitions) is already a list (a ';' separated string of flags to be removed).
The string REPLACE function currently causes the ';' list separators to be removed after which remove_definitions() does not work for multiple flags.

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): Master

jpfr, 2019-06-28:
@Pro , your call. 😄 

A possible workaround.
Can you just append the "inverse definition" at the end of the definitions?
Such as `-Wnoerror` to override `-Werror`.

gbeuning, 2019-06-28:
That could be a workaround in some cases. However if you need to (cross)compile using an older compiler that is unaware of certain flags it typically also doesn't know the 'inverse' flags...
Just for your info, following seems to work properly (CMake version 3.5.1 on Ubuntu 16.04):
```
GET_PROPERTY(ua_architecture_remove_definitions GLOBAL PROPERTY UA_ARCHITECTURE_REMOVE_DEFINITIONS)
if (NOT "${ua_architecture_remove_definitions}" STREQUAL "")
  remove_definitions(${ua_architecture_remove_definitions})
endif(NOT "${ua_architecture_remove_definitions}" STREQUAL "")
```
So `remove_definitions()` seems to work also with a list input.

jpfr, 2021-09-23:
We now have a new way to test and add compiler flags.

closed 2021-09-23 by jpfr