open62541/open62541 #2857: Compile Error of OPEN62541

xiaotian711, 2019-06-30:
[image]
[image]

## Description   through using VS2017,   Trying to connect to UA server, show error:No suitable UserTokenPolicy found for the possible endpoints!

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Stasik0, 2019-06-30:
It is not a compile error, is the server offering nosecurity endpoint?

Pro, 2019-07-01:
You are probably trying to connect to a server, which does not provide an endpoint for username/password authentication. 

Since you did not fill out the issue template completely, there's essential information missing. I'll close this as unclear question.

Please extend your post with the necessary information, then we can have another look.

closed 2019-07-01 by Pro