open62541/open62541 #2860: PubSub packet latency

canadyre, 2019-07-01:
## Description
Hi, I am trying to calculate the latency of pub/sub packets on Raspberry Pis that are connected via a router. I am sending the time that the publisher sends the message in the packet and then getting the time the packet was received at the subscriber end. However, I am getting a latency value of around 30 ms which seems about an order of magnitude too high. Has anyone ever tried to do this before or are there any tips on how to do this?

Thanks!
## Background Information / Reproduction Steps

Used CMake options:
UA_ENABLE_PUBSUB ON
UA_ENABLE_PUBSUB_ETH_UADP ON
UA_BUILD_EXAMPLES ON
CMAKE_C_COMPILER /usr/bin/arm-linux-gnueabihf-gcc

## Checklist

Please provide the following information:

 - [0.30 ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ Raspbian] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

ccvca, 2019-07-02:
So your 30ms latency is calculated on the receiver side, by subtracting the current time with the time in the package?
In this case please evaluate that the clocks of the raspberry Pis are synchronized down to ms range. Probably they have a difference of some ms.

Pro, 2019-07-02:
As @ccvca already mentions, you have to make sure that the clocks are synchronized. You can use PTP for that.

Also, you have to make sure that the router is routing your packages with high priority (depends on your router), otherwise you may end up with jitter.

Maybe this may also be interesting for you: https://www.kernel.org/doc/Documentation/networking/timestamping.txt

kingalexsylvester, 2019-08-05:
The complicated procedure:

1. PTP Sync (only software timestamp is possible on RPi3 and you need to do PTP specific changes to kernel and rebuild, 3hrs procedure)
2. PHC2SYS sync is needed
3. compensate for the poor accuracy due to non RT environment and software clock

Easier path:

easier measurement will be to send 1 byte to  
node 2 and receive the same data at node 1 and try to find the overall time required for this and simply divide by 2 to get your latency. In this method you will be doing the measurement in the same time context/domain which removes you the headache of Ptp and PHC2SYS.

My RPi3 showed far better performance than yours even though I had a legacy Ethernet switch in between.

Please refer the issue #2486 and share your feedback

teccheck, 2025-02-25:
The given answers look good to me and there hasn't been any activity for a long time, so I'll close this issue

closed 2025-02-25 by teccheck