open62541/open62541 #2862: Upload file from UAExpert client to the Server on STM32

Feras992, 2019-07-02:
Hello there,

i looked in the example folder for an example on how to upload file "binary file for example" from the client "UAExpert" to a server runs on STM32F429.

is there any example for it? support open62541 File uploading?

## Checklist

Please provide the following information:

 - [*] open62541 Version (release number or git tag): FixFreeRTOS Branch
 - [*] Other OPC UA SDKs used (client or server): UAExpert
 - [*] Operating system: FreeRTOS
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

embbo, 2019-07-02:
I have an idea... if you make call back function and make "string" as input arugument then you can upload whole string as input arguemnt .

jpfr, 2019-07-03:
Umm. What binary do you mean?
The server binary you want to run?

Feras992, 2019-07-03:
@embbo in Specification part 5 annex c there is somthing called Temporary File Transfer so i think it should be the way how to upload a file but i couldn't find any implementation for it.

@jpfr i mean with it a binary file like .bin .. it can be the binary code of STM32 so i can update the Software over the opc ua

jpfr, 2019-07-03:
You should not use OPC UA to update the firmware.
It was made for runtime communication. Not for development purposes.

Ask your vendor about the compiler and toolchain he recommends.

jpfr, 2019-07-03:
Or is this a feature where the end-customer can flash the firmware?

Feras992, 2019-07-05:
> 
> 
> Or is this a feature where the end-customer can flash the firmware?

@jpfr  yes it will be a feature to flash a new firmware using opc ua. i'm not sure if open62541 support somthing like this

schroeder-, 2019-07-05:
There is a FileType which you can use, but you have to implement your own instance. There is no out of the box support for it.

Feras992, 2019-07-05:
@schroeder-  i couldn't find the FileType in open62541.
do u mean that i have to implement the whole FileType with all his Variables and Methods like "open, read, write, etc ... )?

schroeder-, 2019-07-05:
> @schroeder- i couldn't find the FileType in open62541.
You have to build with CMake option namespace=full or you add **FileType** to tools/schema/datatypes_minimal.txt and rebuild with cmake. 

> do u mean that i have to implement the whole FileType with all his Variables and Methods like "open, read, write, etc ... )?
you have to instantiate a object from FileType and implement the OPC UA Methods Close, GetPostion, Open, Read, SetPosition, Write and some properties.

Pro, 2019-08-30:
I guess the main issue is answered. If not, please reopen.

closed 2019-08-30 by Pro

EmbSys2019, 2020-05-11:
@schroeder- i have the same problem, i tried to add the FileType to the datatypes_minimal.txt and rebuild the stack using cmake.

i'm not pretty sure how to instantiate an object from the FileType.

`
  UA_NodeId fileID;
  UA_ObjectAttributes fileAttr = UA_ObjectAttributes_default;
  fileAttr.displayName = UA_LOCALIZEDTEXT("en-US", "File");
  UA_Server_addObjectNode(server, UA_NODEID_NULL,
                          UA_NODEID_NUMERIC(0, UA_NS0ID_FILETYPE),
                          UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES),
                          UA_QUALIFIEDNAME(1, "File"), UA_NODEID_NUMERIC(0, UA_NS0ID_BASEOBJECTTYPE),
                          fileAttr, NULL, &fileID);
`

Feras992, 2020-06-02:
hello and sorry for opening the issue once again

i have a problem with the FileType .. first of all i tried to add the FileTypt to the minimal Server as @schroeder- wrote, but it didn't work. so i copied the functions from the full server to the minimal server and added the FileType manually.

i started to implement the functions of the FileType(very basic implementation) and i instantiated an objcet from the FileType.
i instantiate the object using this code:

```
  UA_ObjectAttributes fileAttr = UA_ObjectAttributes_default;
  fileAttr.displayName = UA_LOCALIZEDTEXT("en-US", "Firmware");
  UA_Server_addObjectNode(server,
                          UA_NODEID_NUMERIC(1, 11575),
                          UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
                          UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES),
                          UA_QUALIFIEDNAME(1, "Firmware"),
                          UA_NODEID_NUMERIC(0, UA_NS0ID_FILETYPE),
                          fileAttr,
                          NULL,
                          NULL);
```
and i can see the object Firmware in the Server:

[image]

and i choosed write from local file and choosed the file to be sent and i get this error:
[image]

after that i tried to use every method in the FileType(read and write ...) and they worked
[image]

i wrote a text file from the laptop to the server and sent it back over UART and i saved a massage from the uC to the laptop over the OPCUA server.

i didn't used the attributes from the FileType(Size, Writable, MimeType ...) in my code and i couldn't know to use them in this case.

Any help please? do i instantiate the object correctly? how to use the attributes?

thnx

samAvr, 2021-07-14:
@Feras992 Have you solved it?

@jpfr i used this PR: [[WIP] FileType implementation](https://github.com/open62541/open62541/pull/4436/commits/3a5743269bd169e5d4cdcf5c69e5b19187682660) and it works when i upload a small file, which fits in my free ram. the problem is, i want know to upload a file (about 40Kb) and the free available RAM is about 25Kb.
The write callback function receives the whole data from the file. is there any way to receive small pieces of the file and handle it?