open62541/open62541 #2863: Historical data example code

GLmontanari, 2019-07-02:
## Description
Hello! I tried to run the two examples tutorial_server_historicaldata.c and client_historical.c in my
MSVS2017 environment. I changed nothing in the code, just copy-pasted it. It compiles and run but when I try to read the only node in the server i get
```
Start historical read (1, "myUintValue"):
Failed. BadHistoryOperationUnsupported
```

Because the call to 
```
	 UA_NodeId node = UA_NODEID_STRING(1, (char*)"myUintValue");
	 retval = UA_Client_HistoryRead_raw(client, &node, readHist,
		 UA_DateTime_fromUnixTime(0), UA_DateTime_now(), UA_STRING_NULL, false, 10, UA_TIMESTAMPSTORETURN_BOTH, (void *)UA_FALSE);
```

returned retval with such status code.
Where did I go wrong? 

## Background Information / Reproduction Steps
I also tried to read the node using UA Expert but I got the same error prompt.

The examples where built using MSVS2017 on Windows 10. Open62541 version 0.3.
Win32 + Release mode.
I built the library using only the default options which one finds when open Cmake. amalgamation was unchecked.

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 0.3
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: windows10
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

juangburgos, 2019-08-08:
Try with latest master. Compile using CMake option `UA_ENABLE_HISTORIZING` set to true. I tried with VS2017 and UaExpert and it works.

teccheck, 2025-02-25:
If you compile with `UA_ENABLE_HISTORIZING`, as @juangburgos wrote, it should work. It's been quite some time since the last response here so I'll close this issue

closed 2025-02-25 by teccheck