open62541/open62541 #2864: Need for better TCP and mDNS interface selection

Pro, 2019-07-02:
As already described here:
https://github.com/open62541/open62541/pull/2856#issuecomment-507580126

Currently we are just adding a tcp socket for every interface we find:
https://github.com/open62541/open62541/blob/master/arch/network_tcp.c#L359
In the future this should be changed to give the user an option to select a specific interface by specifying an IP address where the server should listen (i.e., 0.0.0.0 is the default and means all interfaces). That should then also be part of #2408 here:
https://github.com/open62541/open62541/pull/2408/files#diff-431422d938c278cd653aa380ecdcfd40R292 (ping @mlgiraud)

For mDNS we could use this information and if the tcp listener interface is set to:
- 0.0.0.0:
	- default: check all interfaces which have multicast enabled, and send mDNS on all of them
    - in addition the user may limit the interface selection to a subset of these
- specific ip: only send mDNS packages on the interface which has that ip

Does that make sense?

The next issue is then, what about the case where a server only has UDP interface, or even ETH?
Maybe the most intuitive configuration option is, to configure the mDNS socket independent of the listener sockets.

I.e.:
- Default behaviour: check all interfaces which have multicast enabled, and send mDNS on all of them
- The user can select one or more interfaces where mDNS is sent

mlgiraud, 2019-07-02:
I would like to get the PR merged before making further changes.
I'm currently working on fixing some last issues on windows.

Changing the code to make specifying an interface possible should be pretty straightforward, as we only need to change the function that creates the socket.

Pro, 2019-07-02:
Yes, definitely. We should try to get #2408 in as soon as possible and then create follow-up PRs.

This is just a meta issue/reminder/idea collection to implement that feature.