open62541/open62541 #2865: OPC UA Pub/Sub over MQTT

canadyre, 2019-07-02:
## Description
Hi all,

I am trying to work with the OPC UA PubSub with MQTT on a couple of Raspberry Pis. I was wondering if there were any examples of this that anyone has come across? The only examples I can find use UDP over UADP. I also read an issue that said that AMQP was not supported even though it mentions in the Documentation so is that the case with MQTT as well? Any help would be greatly appreciated.

Thanks!

## Background Information / Reproduction Steps

Used CMake options:
UA_ENABLE_PUBSUB ON
UA_ENABLE_PUBSUB_ETH_UADP ON
UA_BUILD_EXAMPLES ON
CMAKE_C_COMPILER /usr/bin/arm-linux-gnueabihf-gcc

## Checklist

Please provide the following information:

 - [0.3 ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ Raspbian] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

theresa95, 2019-07-03:
As far as I know, the PubSub mechanism is currently only supported by UADP.

sohailmushtaq, 2019-07-03:
Hi, 
there is a branch which contains OPC UA PubSub MQTT with examples also. Check the branch pubsub_mqtt_clean. My master thesis is about OPC UA PubSub, so I've worked on PubSub with MQTT, in case you have questions just let me know.

Regards

Sohail

canadyre, 2019-07-03:
Hi,
Thank you for helping me find that @sohailmushtaq . So when running these examples with mqtt, do you still run tutorial_pubsub_publish, tutorial_pubsub_subscribe and then use tutorial_pubsub_mqtt as the broker? 
Thanks!

sohailmushtaq, 2019-07-03:
Hi,
No, use only tutotial_pubsub_mqtt. First of all install a MQTT broker, such as Mosquitto (it's easy to install and use, I reccomend that). After you the Broker is running, just execute pubsub_mqtt you should see packets sent from Publisher to Broker on Wireshark. If you want to receive as well you have to configure also the Subscriber. Once both configured and connected to Broker, Publisher will send information through Broker to Subscriber.

jpfr, 2019-07-04:
We have a current PR for pubsub with MQTT in #2796.

rudoq32, 2019-07-16:
Hello @sohailmushtaq,
I am trying to use tutorial_pubsub_mqtt. It seems that when I compile the project with UA_BUILD_EXAMPLES it is not compiled.
I am struggling to create an C project in MS VC with tutorial_pubsub_mqtt. 
By a chance do you have a repository where is tutorial_pubsub_mqtt built as an example?

sohailmushtaq, 2019-07-16:
Hi @rudoq32,
I'm not familiar with MS VC. I hope I'll be helpful even if I've worked on Ubuntu. When you compile the project you don't need only UA_BUILD_EXAMPLES but also the advanced options: UA_ENABLE_PUBSUB ON, UA_ENABLE_PUBSUB_ETH_UADP ON, etc. Including UA_ENABLE_PUBSUB_MQTT which enables the MQTT feature. Check if these options are all enabled and check that in the CmakeLists.txt in the folder examples contains this: 
if(UA_ENABLE_PUBSUB_MQTT)
        add_example(tutorial_pubsub_mqtt pubsub/tutorial_pubsub_mqtt.c)
endif()
Finally check also that UA_ENABLE_JSON_ENCODING is enabled for JSON encoding. After all these checks it should work. Let me know if you have any problems.

Regards,

Sohail

rudoq32, 2019-07-17:
Hello @sohailmushtaq,
thank you for the answer. I see an issue with UA_ENABLE_PUBSUB_ETH_UADP . For the moment, as far as I see, UA_ENABLE_PUBSUB_ETH_UADP is available only for linux.
I am planing to use the code with freertos. But as first, I tried to use it on windows. So I guess I have a long journey to test it :)

kusumy, 2019-08-29:
hi @sohailmushtaq, i am interested on your OPC UA thesis, because i am also doing a thesis for opc ua pub sub implementation for Battery Management System. I hope you can share your thesis, so i can have more information about opc ua pubsub.

sohailmushtaq, 2019-08-29:
Hi @kusumy, currently I'm writing the thesis it will be ready not before the first week of October. You can contact me and I will send you my final report. 

Regards,

Sohail

kusumy, 2019-09-03:
@sohailmushtaq, is there any email address or other social media account so i can contact you?

Regards

sohailmushtaq, 2019-09-04:
Hi @kusumy , you can contact me at the following email address sohail95@live.it
Regards
Sohail

kusumy, 2019-09-27:
hi @sohailmushtaq , 
Did you receive my email? 

Regards

kusumy, 2019-09-27:
> Hi,
> there is a branch which contains OPC UA PubSub MQTT with examples also. Check the branch pubsub_mqtt_clean. My master thesis is about OPC UA PubSub, so I've worked on PubSub with MQTT, in case you have questions just let me know.
> 
> Regards
> 
> Sohail

Hi @sohailmushtaq  can you show me where this branch is ?

Regards

kusumy, 2019-09-27:
> Hi @sohailmushtaq can you show me where this branch is ?
>
> Regards

Please ignore @sohailmushtaq , i have found it 

Regards

RavilN, 2019-11-21:
Hi, 
Cannot find that branch pubsub_mqtt_clean, was it deleted? If so, was the code merged into master?

ramadan-shweiki, 2020-10-29:
Hello @sohailmushtaq @kusumy :)

I am also currently writing my master thesis with open62541 Pubsub brokerless variant. I have the tutorial_pubsub_publish.c and pubsub_subscribe_subscribe but unfortunately the pubsub_subscribe does not get anything because it is not fully implemented.

I don't know if it is possible to continue working with this library I need the PubSub brokerless variant.
when i run the tutorial_pubsub_standalone then i get the published messages, but i am sceptical if it is possible to continue working with the class and publish other values.

I hope that you can share your thesis, so I have also more experiences and information about opc ua pubsub.

My thesis is about using opc ua pubsub for in-vehicle communication (communication between sensors and actuators for a use case in vehicle).

you can contact me at the following email address: rs.shweiki@gmail.com

best Regards

teccheck, 2025-02-25:
Examples for pubsub over MQTT exist now, and they should work on an RPI too, as it's also Debian (at least in most cases)

closed 2025-02-25 by teccheck