open62541/open62541 #2866: Cannot add more than one reference (different types) between two objects

juangburgos, 2019-07-04:
## Description

I create two different NonHierarchical reference types `ref1Type` and `ref2Type`, then I create two object instances `obj1` and `obj2`. I add both references types from `obj1` to `obj2`, only the first one works:

```
obj1 ----- ref1Type ----> obj2
obj1 ----- ref2Type ----> obj2
```

The result from `UA_Server_addReference` succeeds in both cases, but in UAExpert, only the first one added is visible. Also when using the server's browse service, it fails to find the second reference. Furthermore, if try to add again the second one, it claims to be repeated.

I attach code to reproduce.

## Background Information / Reproduction Steps

```
#include <assert.h>
#include <signal.h>
#include "open62541.h"

static UA_NodeId findReference(const UA_NodeId sourceId, const UA_NodeId refTypeId, const bool isForward, UA_Server * server)
{
	UA_NodeId outNodeId = UA_NODEID_NULL;
	// make ua browse
	UA_BrowseDescription * bDesc = UA_BrowseDescription_new();
	UA_NodeId_copy(&sourceId, &bDesc->nodeId);
	bDesc->browseDirection = isForward ? UA_BROWSEDIRECTION_FORWARD : UA_BROWSEDIRECTION_INVERSE;
	bDesc->includeSubtypes = true;
	bDesc->resultMask = UA_BROWSERESULTMASK_REFERENCETYPEID;
	// browse
	UA_BrowseResult bRes = UA_Server_browse(server, 0, bDesc);
	assert(bRes.statusCode == UA_STATUSCODE_GOOD);
	while (bRes.referencesSize > 0)
	{
		for (size_t i = 0; i < bRes.referencesSize; i++)
		{
			UA_ReferenceDescription rDesc = bRes.references[i];
			if (UA_NodeId_equal(&rDesc.referenceTypeId, &refTypeId))
			{
				outNodeId = rDesc.nodeId.nodeId;
				break;
			}
		}
		UA_BrowseResult_deleteMembers(&bRes);
		bRes = UA_Server_browseNext(server, true, &bRes.continuationPoint);
	}
	// cleanup
	UA_BrowseDescription_deleteMembers(bDesc);
	UA_BrowseDescription_delete(bDesc);
	UA_BrowseResult_deleteMembers(&bRes);
	// return
	return outNodeId;
}

static UA_NodeId registerRefType(const char * forwName, const char * invName, UA_Server * server)
{
	UA_NodeId outNodeId;
	UA_QualifiedName browseName;
	browseName.namespaceIndex = 1;
	browseName.name = UA_STRING_ALLOC(forwName);
	// setup new ref type attributes
	UA_ReferenceTypeAttributes refattr = UA_ReferenceTypeAttributes_default;
	refattr.displayName = UA_LOCALIZEDTEXT((char*)(""), forwName);
	refattr.inverseName = UA_LOCALIZEDTEXT((char*)(""), invName );
	UA_StatusCode st = UA_Server_addReferenceTypeNode(
		server,
		UA_NODEID_NULL,
		UA_NODEID_NUMERIC(0, UA_NS0ID_NONHIERARCHICALREFERENCES),
		UA_NODEID_NUMERIC(0, UA_NS0ID_HASSUBTYPE),
		browseName,
		refattr,
		NULL,
		&outNodeId
	);
	assert(st == UA_STATUSCODE_GOOD);
	// clean up
	UA_QualifiedName_clear(&browseName);
	return outNodeId;
}

static UA_NodeId addObjInstance(const UA_NodeId parentNodeId, const char * dispName, UA_Server * server)
{
	UA_NodeId outNodeId;
	UA_ObjectAttributes oAttr = UA_ObjectAttributes_default;
	oAttr.displayName = UA_LOCALIZEDTEXT("en-US", dispName);
	// add variable
	UA_QualifiedName browseName;
	browseName.namespaceIndex = 1;
	browseName.name = UA_STRING_ALLOC(dispName);
	// add instance
	UA_StatusCode st = UA_Server_addObjectNode(server,
		UA_NODEID_NULL,                                // reuested nodeid
		parentNodeId,                                  // parent
		UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT),   // parent relation with child
		browseName,
		UA_NODEID_NUMERIC(0, UA_NS0ID_BASEOBJECTTYPE), // type
		oAttr,
		(void*)"MyObjectInstance",                     // context
		&outNodeId);                      // new variable instance id
	assert(st == UA_STATUSCODE_GOOD);
	return outNodeId;
}

UA_Boolean running = true;
static void stopHandler(int sign)
{
	UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "received ctrl-c");
	running = false;
}

int main(void)
{

	UA_Server *server = UA_Server_new();
	UA_ServerConfig_setDefault(UA_Server_getConfig(server));

	UA_NodeId objectsNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);
	// create two different reference types
	UA_NodeId ref1TypeId = registerRefType("HasRef1", "IsRefOf1", server);
	UA_NodeId ref2TypeId = registerRefType("HasRef2", "IsRefOf2", server);
	// create two different object instances
	UA_NodeId sourceId = addObjInstance(objectsNodeId, "obj1", server);
	UA_NodeId targetId = addObjInstance(objectsNodeId, "obj2", server);
	// connect them twice, one time per reference type
	UA_ExpandedNodeId targetExpId;
	targetExpId.nodeId       = targetId;
	targetExpId.namespaceUri = UA_STRING_NULL;
	targetExpId.serverIndex  = 0;
	UA_StatusCode st;
	st = UA_Server_addReference(server, sourceId, ref1TypeId, targetExpId, true);
	assert(st == UA_STATUSCODE_GOOD);
	st = UA_Server_addReference(server, sourceId, ref2TypeId, targetExpId, true);
	assert(st == UA_STATUSCODE_GOOD);
	// check references where added
	UA_NodeId targetCheckId;
	targetCheckId = findReference(sourceId, ref1TypeId, true, server);
	assert(UA_NodeId_equal(&targetCheckId, &targetId));
	// BELOW FAILS !
	targetCheckId = findReference(sourceId, ref2TypeId, true, server);
	assert(UA_NodeId_equal(&targetCheckId, &targetId));
	// ALSO FAILS, CLAIMING TO BE REPEATED !
	st = UA_Server_addReference(server, sourceId, ref2TypeId, targetExpId, true);
	assert(st == UA_STATUSCODE_GOOD);

	UA_StatusCode retval = UA_Server_run(server, &running);
	UA_Server_delete(server);
	return (int)retval;
}
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): master
 - [x] Operating system: Windows
 - [x] Self-contained code example attached
 - [x] Critical issue

juangburgos, 2019-07-05:
I have beed using the debugger to try and find out where it goes wrong and the issue seems to be at the beginning of the `browseNodeRefKind` function, where there is a condition to check if the target is already added to the browse result:

```c
static UA_StatusCode UA_FUNC_ATTR_WARN_UNUSED_RESULT
browseNodeRefKind(UA_Server *server, UA_Session *session, ContinuationPoint *cp,
                  RefResult *rr, UA_Boolean *maxed, const UA_NodeReferenceKind *rk,
                  const UA_ExpandedNodeId *target) {
    /* Is the target already in the tree? */
    RefEntry re;
    re.target = &target->nodeId;
    re.targetHash = UA_NodeId_hash(&target->nodeId);
    if(ZIP_FIND(RefHead, &cp->rt.head, &re)) {
        cp->nki++;
        return UA_STATUSCODE_GOOD;
    }

// ... other tuff

}
```

Is this a correct thing to do? Shouldn't it also check the reference type and not only the target to mark it as a copy? Also, what would be the correct way to fix it? I tried following the code, but find it difficult due to the macro wizardry.

jpfr, 2019-07-08:
Thanks for the detailed report.
I converted the code example to a new unit test.
The fix is in #2874.

PR #2874 (2019-07-31): Refactor browse to fix #2866
fixes #2866
closed 2019-07-31 by jpfr