open62541/open62541 #2867: Logging to file

jopi2016, 2019-07-05:
## Description
Hi,
(I went through older issues but couldn't find anything related to my question).
I would like to log instead of to stdout into a file. The only plugin I can find in the plugins folder is ua_log_stdout. Do there already exist other plugins?

Is there somewhere a tutorial about how to setup open62541 for logging to a file?

Thanks
jopi2016

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jopi2016, 2019-07-05:
well,
I already found a solution / workaround.

One option is to redirect all output to stdout and stderror into a file.

`  freopen( "output.txt", "w", stdout );
  freopen( "error.txt", "w", stderr );`

see [https://stackoverflow.com/a/7400528](stackoverflow)

schroeder-, 2019-07-05:
You can replace the UA_Logger see UA_ServerConfig.logger orUA_ClientUA_ClientConfig.logger with you own instance see [https://open62541.org/doc/current/plugin_log.html?highlight=ua_logger](url).

jopi2016, 2019-07-05:
thank you for your quick answer - I'll give it a try.

closed 2019-07-05 by jopi2016