open62541/open62541 #2868: Compiling a complicated NodeSet

miehe-dup, 2019-07-05:
## Description

This is a follow-up on issue #2784 -- I still can't compile my Nodeset.

Namespaces:

- NS0: http://opcfoundation.org/UA/
- NS1: http://opcfoundation.org/UA/DI/
- NS2: http://www.euromap.org/euromap83/
- NS3: http://www.euromap.org/euromap82_2/
- NS4: (vendor specific)

Files I prepared:

- Opc.Ua.NodeSet2.xml
- Opc.Ua.Di.NodeSet2.xml
- Opc.Ua.Di.NodeSet2.csv
- Opc.Ua.Di.Types.bsd
- Opc_Ua.EUROMAP83.1_00_ext82_2_RC1_00_1.NodeSet2.xml
- Opc_Ua.EUROMAP83.1_00_ext82_2_RC1_00_1.NodeSet2.csv
- Opc_Ua.EUROMAP83.1_00_ext82_2_RC1_00_1.Types.bsd
- Opc_Ua.EUROMAP82_2.RC1_00_1.NodeSet2.xml
- Opc_Ua.EUROMAP82_2.RC1_00_1.NodeSet2.csv
- Opc_Ua.EUROMAP82_2.RC1_00_1.Types.bsd
- (my files)

## Background Information / Reproduction Steps

Following documentation at https://open62541.org/doc/0.3/nodeset_compiler.html#combination-of-multiple-nodesets -- I tried this. 

Error output included.

```
$ open62541/generate_datatypes.py --namespace=1 --type-csv=../model/Opc.Ua.Di.NodeSet2.csv --type-bsd=../model/Opc.Ua.Di.Types.bsd --no-builtin open62541/ua_types_di
$ open62541/generate_datatypes.py --namespace=2 --type-csv=../model/Opc_Ua.EUROMAP83.1_00_ext82_2_RC1_00_1.NodeSet2.csv --type-bsd=../model/Opc_Ua.EUROMAP83.1_00_ext82_2_RC1_00_1.Types.bsd --no-builtin open62541/ua_types_eu83
$ open62541/generate_datatypes.py --namespace=3 --type-csv=../model/Opc_Ua.EUROMAP82_2.RC1_00_1.NodeSet2.csv --type-bsd=../model/Opc_Ua.EUROMAP82_2.RC1_00_1.Types.bsd --no-builtin open62541/ua_types_eu82

$ open62541/nodeset_compiler/nodeset_compiler.py --internal-headers --types-array=UA_TYPES --types-array=UA_TYPES_DI --existing ../model/Opc.Ua.NodeSet2.xml --xml ../model/Opc.Ua.Di.NodeSet2.xml open62541/ua_namespace_di
INFO:__main__:Preprocessing (existing) ../model/Opc.Ua.NodeSet2.xml
INFO:__main__:Preprocessing ../model/Opc.Ua.Di.NodeSet2.xml
INFO:__main__:Generating Code
INFO:__main__:NodeSet generation code successfully printed

$ open62541/nodeset_compiler/nodeset_compiler.py --internal-headers --types-array=UA_TYPES --types-array=UA_TYPES_DI --types-array=UA_TYPES_EU83 --existing ../model/Opc.Ua.NodeSet2.xml --existing ../model/Opc.Ua.Di.NodeSet2.xml --xml ../model/Opc_Ua.EUROMAP83.1_00_ext82_2_RC1_00_1.NodeSet2.xml open62541/ua_namespace_eu83
INFO:__main__:Preprocessing (existing) ../model/Opc.Ua.NodeSet2.xml
INFO:__main__:Preprocessing (existing) ../model/Opc.Ua.Di.NodeSet2.xml
INFO:__main__:Preprocessing ../model/Opc_Ua.EUROMAP83.1_00_ext82_2_RC1_00_1.NodeSet2.xml
Traceback (most recent call last):
  File "open62541/nodeset_compiler/nodeset_compiler.py", line 178, in <module>
    ns.allocateVariables()
  File "open62541/nodeset_compiler/nodeset.py", line 322, in allocateVariables
    n.allocateValue(self)
  File "open62541/nodeset_compiler/nodes.py", line 291, in allocateValue
    self.value.parseXMLEncoding(self.xmlValueDef, dataTypeNode, self)
  File "open62541/nodeset_compiler/datatypes.py", line 167, in parseXMLEncoding
    self.value = [self.__parseXMLSingleValue(xmlvalue, parentDataTypeNode, parent)]
  File "open62541/nodeset_compiler/datatypes.py", line 206, in __parseXMLSingleValue
    t.parseXML(xmlvalue)
  File "open62541/nodeset_compiler/datatypes.py", line 572, in parseXML
    if not self.stringRepresentation == xmlvalue.localName:
AttributeError: 'NodeId' object has no attribute 'stringRepresentation'
```

This archive includes all the aforementioned files: [model.zip](https://github.com/open62541/open62541/files/3362661/model.zip)

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): 0.3.0
 - [X] Other OPC UA SDKs used (client or server): none
 - [X] Operating system: Linux
 - [X] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [X] Self-contained code example attached
 - [ ] Critical issue

miehe-dup, 2019-07-11:
A `git bisect` using "stringRepresentation" (referring to the `AttributeError: 'NodeId' object has no attribute 'stringRepresentation'`) brought up commit 7bd3fd876f854a4e5fdd53bcb2094ee823b39a1b fixing issue #1823. 

I'm a tad bit confused now as to why a commit from June 2018 is not included in release 0.3.0, but it led me to try the master branch version of the nodeset compiler. This turned out to compile my everything up to NS3 flawlessly! (Our NS4 is highly unstable right now, so not yet included in our build process)

```
$ open62541/generate_datatypes.py --namespace=2 --type-csv=../model/Opc_Ua.EUROMAP83.1_00_ext82_2_RC1_00_1.NodeSet2.csv --type-bsd=../model/Opc_Ua.EUROMAP83.1_00_ext82_2_RC1_00_1.Types.bsd --no-builtin open62541/ua_types_eu83
$ open62541/generate_datatypes.py --namespace=3 --type-csv=../model/Opc_Ua.EUROMAP82_2.RC1_00_1.NodeSet2.csv --type-bsd=../model/Opc_Ua.EUROMAP82_2.RC1_00_1.Types.bsd --no-builtin open62541/ua_types_eu82
$ open62541/nodeset_compiler/nodeset_compiler.py --internal-headers --types-array=UA_TYPES --types-array=UA_TYPES_DI --existing ../model/Opc.Ua.NodeSet2.xml --xml ../model/Opc.Ua.Di.NodeSet2.xml open62541/ua_namespace_di
INFO:__main__:Preprocessing (existing) ../model/Opc.Ua.NodeSet2.xml
INFO:__main__:Preprocessing ../model/Opc.Ua.Di.NodeSet2.xml
INFO:__main__:Generating Code for Backend: open62541
INFO:__main__:NodeSet generation code successfully printed
$ open62541/nodeset_compiler/nodeset_compiler.py --internal-headers --types-array=UA_TYPES --types-array=UA_TYPES_DI --types-array=UA_TYPES_EU83 --existing ../model/Opc.Ua.NodeSet2.xml --existing ../model/Opc.Ua.Di.NodeSet2.xml --xml ../model/Opc_Ua.EUROMAP83.1_00_ext82_2_RC1_00_1.NodeSet2.xml open62541/ua_namespace_eu83
INFO:__main__:Preprocessing (existing) ../model/Opc.Ua.NodeSet2.xml
INFO:__main__:Preprocessing (existing) ../model/Opc.Ua.Di.NodeSet2.xml
INFO:__main__:Preprocessing ../model/Opc_Ua.EUROMAP83.1_00_ext82_2_RC1_00_1.NodeSet2.xml
INFO:__main__:Generating Code for Backend: open62541
INFO:__main__:NodeSet generation code successfully printed
$ open62541/nodeset_compiler/nodeset_compiler.py --internal-headers --types-array=UA_TYPES --types-array=UA_TYPES_DI --types-array=UA_TYPES_EU83 --types-array=UA_TYPES_EU82 --existing ../model/Opc.Ua.NodeSet2.xml --existing ../model/Opc.Ua.Di.NodeSet2.xml --existing ../model/Opc_Ua.EUROMAP83.1_00_ext82_2_RC1_00_1.NodeSet2.xml --xml ../model/Opc_Ua.EUROMAP82_2.RC1_00_1.NodeSet2.xml open62541/ua_namespace_eu82
INFO:__main__:Preprocessing (existing) ../model/Opc.Ua.NodeSet2.xml
INFO:__main__:Preprocessing (existing) ../model/Opc.Ua.Di.NodeSet2.xml
INFO:__main__:Preprocessing (existing) ../model/Opc_Ua.EUROMAP83.1_00_ext82_2_RC1_00_1.NodeSet2.xml
INFO:__main__:Preprocessing ../model/Opc_Ua.EUROMAP82_2.RC1_00_1.NodeSet2.xml
INFO:__main__:Generating Code for Backend: open62541
INFO:__main__:NodeSet generation code successfully printed
```

Pro, 2019-07-11:
The nodeset compiler for master changed quite a bit and is using the newer API from master.
Therefore you will have no success on using the NC output from master in 0.3.

Also, on the master branch the nodeset compiler contains many bugfixes and improvements. It's also tested with some EUROMAP Nodesets.

I currently do not have the ressources to backport these improvements to 0.3, and since the next release is quite near, this additional work does not really make sense.

Therefore I'd recommend that you use master. Would that be an option for you?

miehe-dup, 2019-07-11:
Sounds good, but our current setup uses the single file amalgamate distribution (this integrates into our yocto build process rather nicely).

Is there a way to get a single-file version of the master branch?

Pro, 2019-07-11:
Yes, definitely.
`cmake -D UA_ENABLE_AMALGAMATION=ON ..` should still work the same way as on 0.3 and produce the single file distribution, i.e. open62541.c/.h

miehe-dup, 2019-07-12:
That worked, thank you @Pro! However, new day, new problem.
I had setup a simple demo server for ourself (which was really straightforward to port to 0.4.0-master btw, good job on a mostly consistent API!).
Now, I tried adding just the DI namespace into that:

```
$ open62541/generate_datatypes.py --namespace=1 --type-csv=../model/Opc.Ua.Di.NodeSet2.csv --type-bsd=../model/Opc.Ua.Di.Types.bsd --no-builtin open62541/types_di
$ open62541/nodeset_compiler/nodeset_compiler.py --internal-headers --types-array=UA_TYPES --types-array=UA_TYPES_DI --existing ../model/Opc.Ua.NodeSet2.xml --xml ../model/Opc.Ua.Di.NodeSet2.xml open62541/namespace_di
INFO:__main__:Preprocessing (existing) ../model/Opc.Ua.NodeSet2.xml
INFO:__main__:Preprocessing ../model/Opc.Ua.Di.NodeSet2.xml
INFO:__main__:Generating Code for Backend: open62541
INFO:__main__:NodeSet generation code successfully printed
open62541/nodeset_compiler/nodeset_compiler.py --internal-headers   --existin  8,64s user 0,37s system 69% cpu 13,004 total
```
So nodeset compilation seems fine. Added the `types_di_generated` and `namespace_di` as dependency in our Makefile:
```
$ make example_opc
cc -pthread -DUA_ENABLE_AMALGAMATION    -c -o open62541/open62541.o open62541/open62541.c
cc -pthread -DUA_ENABLE_AMALGAMATION    -c -o open62541/types_di_generated.o open62541/types_di_generated.c
cc -pthread -DUA_ENABLE_AMALGAMATION    -c -o open62541/namespace_di.o open62541/namespace_di.c
open62541/namespace_di.c: In function 'function_namespace_di_47_begin':
open62541/namespace_di.c:1358:107: error: 'UA_TYPES_IDTYPE' undeclared (first use in this function); did you mean 'UA_TYPES_INT64'?
 UA_Variant_setArray(&attr.value, &variablenode_ns_1_i_15006_variant_DataContents, (UA_Int32) 1, &UA_TYPES[UA_TYPES_IDTYPE]);
                                                                                                           ^~~~~~~~~~~~~~~
                                                                                                           UA_TYPES_INT64
open62541/namespace_di.c:1358:107: note: each undeclared identifier is reported only once for each function it appears in
<builtin>: recipe for target 'open62541/namespace_di.o' failed
make: *** [open62541/namespace_di.o] Error 1
```
Not as fine :(

So either, the nodeset compilation step hasn't worked as it seemed, or I've used it the wrong way … documentation on the NC has been switched to only cmake in 0.4.0 … my guess is it's not intended to be used like this anymore.

So it seems it's time to port our build to cmake then. To get me set up, suppose I'm starting from scratch just with the "tutorial_server_firststeps.c" and the "open62541" directory (i.e. a git submodule). How could a `CMakeLists.txt` look like then?

miehe-dup, 2019-07-12:
I've come part of the way myself, derived from the `examples/CMakeLists.txt`:

```
# basic setup

cmake_minimum_required(VERSION 3.0...3.12)
project(open62541-examples)
if(${CMAKE_VERSION} VERSION_LESS 3.12)
    cmake_policy(VERSION ${CMAKE_MAJOR_VERSION}.${CMAKE_MINOR_VERSION})
endif()

# open62541 lib

include_directories("${PROJECT_SOURCE_DIR}/open62541")
add_subdirectory (open62541)

# main.c example app

add_executable(main ${STATIC_OBJECTS} main.c)
target_link_libraries(main open62541)
```

I therefore actually created a dir just containing "main.c" (copied code from tutorial_server_firststeps.c) and open62541 master clone. This works.

To test the NC, I added the Ua.Di files in a subdir "model", then had to set some paths and call the cmake macro:
```
# open62541 tools

set (open62541_TOOLS_DIR "${PROJECT_SOURCE_DIR}/open62541/tools")
set (open62541_NODESET_DIR "${PROJECT_SOURCE_DIR}/model")

# ua.di nodeset

ua_generate_nodeset_and_datatypes(
    NAME "di"
    FILE_CSV "${PROJECT_SOURCE_DIR}/model/Opc.Ua.Di.NodeSet2.csv"
    FILE_BSD "${PROJECT_SOURCE_DIR}/model/Opc.Ua.Di.Types.bsd"
    NAMESPACE_IDX 2
    FILE_NS "${PROJECT_SOURCE_DIR}/model/Opc.Ua.Di.NodeSet2.xml"
)
```
However, nothing is generated (although I end up with an empty "src_generated" folder). Am I missing a link?

miehe-dup, 2019-07-12:
Figured most of it out myself by now. Ended up with the same error. Now I start to doubt it's my fault:

```cmake
# [snip unchanged]
# open62541 tools

set (open62541_TOOLS_DIR ${PROJECT_SOURCE_DIR}/open62541/tools)
set (open62541_NODESET_DIR ${PROJECT_SOURCE_DIR}/open62541/deps/ua-nodeset)

# nodeset compiler settings

set (MODEL_GEN_DIR ${PROJECT_BINARY_DIR}/model)
set (MODEL_TARGET_PREFIX "model")

# ua.di nodeset

ua_generate_nodeset_and_datatypes(
    NAME "di"
    NAMESPACE_IDX 1
    
    # our extracted sources
    # FILE_CSV "${PROJECT_SOURCE_DIR}/model/Opc.Ua.Di.NodeSet2.csv"
    # FILE_NS "${PROJECT_SOURCE_DIR}/model/Opc.Ua.Di.NodeSet2.xml"
    # FILE_BSD "${PROJECT_SOURCE_DIR}/model/Opc.Ua.Di.Types.bsd"
    
    # official sources
    FILE_CSV "${open62541_NODESET_DIR}/DI/OpcUaDiModel.csv"
    FILE_NS "${open62541_NODESET_DIR}/DI/Opc.Ua.Di.NodeSet2.xml"
    FILE_BSD "${open62541_NODESET_DIR}/DI/Opc.Ua.Di.Types.bsd"
    
    OUTPUT_DIR "${MODEL_GEN_DIR}"
    TARGET_PREFIX "${MODEL_TARGET_PREFIX}"
)

# ua.di as library (depending on open62541)

add_library(lib_ns-di ${UA_NODESET_DI_SOURCES} ${UA_DI_SOURCES})
target_link_libraries(lib_ns-di open62541)

# link nodeset to main app

target_link_libraries(main lib_ns-di)
```
and the `make` log
```
$ LANG=C make -j
fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
-- The selected architecture is: posix
-- Could NOT find Sphinx (missing: SPHINX_EXECUTABLE) 
-- Could NOT find LATEX (missing: LATEX_COMPILER) 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/entwicklung/Dokumente/opc_test/build
[  1%] Built target open62541-generator-transport
[  3%] Built target open62541-generator-statuscode
[  5%] Built target open62541-generator-types
[  7%] Built target open62541-generator-namespace
[ 28%] Built target open62541-plugins
[ 92%] Built target open62541-object
[ 94%] Built target open62541
[ 94%] Building C object CMakeFiles/lib_ns-di.dir/model/namespace_di_generated.c.o
/home/entwicklung/Dokumente/opc_test/build/model/namespace_di_generated.c: In function 'function_namespace_di_generated_47_begin':
/home/entwicklung/Dokumente/opc_test/build/model/namespace_di_generated.c:1358:107: error: 'UA_TYPES_IDTYPE' undeclared (first use in this function); did you mean 'UA_TYPES_INT64'?
 UA_Variant_setArray(&attr.value, &variablenode_ns_1_i_15006_variant_DataContents, (UA_Int32) 1, &UA_TYPES[UA_TYPES_IDTYPE]);
                                                                                                           ^~~~~~~~~~~~~~~
                                                                                                           UA_TYPES_INT64
/home/entwicklung/Dokumente/opc_test/build/model/namespace_di_generated.c:1358:107: note: each undeclared identifier is reported only once for each function it appears in
CMakeFiles/lib_ns-di.dir/build.make:77: recipe for target 'CMakeFiles/lib_ns-di.dir/model/namespace_di_generated.c.o' failed
make[2]: *** [CMakeFiles/lib_ns-di.dir/model/namespace_di_generated.c.o] Error 1
CMakeFiles/Makefile2:99: recipe for target 'CMakeFiles/lib_ns-di.dir/all' failed
make[1]: *** [CMakeFiles/lib_ns-di.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
```
Sorry for long thread :)

miehe-dup, 2019-07-15:
So the problem is, open62541 must be compiled with the FULL NodeSet 0 to make this work. Add this to the CMakeLists.txt before `add_subdirectory (open62541)`:
```cmake
set (UA_NAMESPACE_ZERO "FULL" CACHE STRING "Use full namespace 0")
```
or create the config with `cmake -DUA_NAMESPACE_ZERO=FULL ..`

closed 2019-07-15 by miehe-dup