open62541/open62541 #2869: HEL/ACK handshake failed

cookiejarlid, 2019-07-05:
## Description

HEL/ACK handshake fails on some opcua servers (eg. opc.tcp://opcua-demo.factry.io:51210) where different clients (eg. uaclient.com) will work.

## Background Information / Reproduction Steps

Simply attempt to connect a client on: opc.tcp://opcua-demo.factry.io:51210

```
[2019-07-05 14:08:45.187 (UTC+0100)] info/client	Connecting to endpoint opc.tcp://opcua-demo.factry.io:51210
[2019-07-05 14:08:45.187 (UTC+0100)] info/client	SecurityPolicy not specified -> use default #None
[2019-07-05 14:08:45.187 (UTC+0100)] warn/securitypolicy	No PKI plugin set. Accepting all certificates
[2019-07-05 14:08:45.243 (UTC+0100)] info/client	TCP connection established
[2019-07-05 14:08:45.297 (UTC+0100)] error/network	Receiving ACK message failed with BadConnectionClosed
[2019-07-05 14:08:45.297 (UTC+0100)] error/client	HEL/ACK handshake failed
[2019-07-05 14:08:45.297 (UTC+0100)] error/client	Couldn't connect the client to a TCP secure channel
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): master branch
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Linux
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [x] Critical issue

mlgiraud, 2019-07-09:
I tried connecting with UAExpert, but the connection is reset by the server:
[image]

cookiejarlid, 2019-07-09:
Thank you. It seems to be hit-and-miss with uaclient.com as well. Closing as it looks like the problem lies in the server.

closed 2019-07-09 by cookiejarlid