open62541/open62541 #2873: Server does not persist the NodeId of a Node

GeorgeSellner, 2019-07-08:
## Description
Server does not persist the NodeId of a Node, that is, it generates new NodeIds when rebooting.

We see this in our server. however, to verify, I checked the examples, and the 'server_inheritance' example exhibits the same behavior.

## Background Information / Reproduction Steps
Linux 4.15.0-54-generic #58-Ubuntu SMP Mon Jun 24 10:55:24 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.2 LTS
Release:	18.04
Codename:	bionic

-- The C compiler identification is GNU 7.4.0
-- The CXX compiler identification is GNU 7.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PythonInterp: /usr/bin/python (found version "2.7.15") 
-- Found Git: /usr/bin/git (found version "2.17.1") 
-- CMAKE_BUILD_TYPE not given; setting to 'Debug'
-- The selected architecture is: posix
-- Performing Test CC_HAS_-fstack-protector-strong
-- Performing Test CC_HAS_-fstack-protector-strong - Success
-- Performing Test CC_HAS_-fstack-clash-protection
-- Performing Test CC_HAS_-fstack-clash-protection - Failed
-- Performing Test CC_HAS_-mcet -fcf-protection
-- Performing Test CC_HAS_-mcet -fcf-protection - Failed
-- Found Sphinx: /usr/bin/sphinx-build  
-- Found LATEX: /usr/bin/latex   
-- Configuring done
-- Generating done
-- Build files have been written to: /home/mmtools/workspace/open62541/build

Used CMake options:
BUILD_SHARED_LIBS:BOOL=OFF
CLANG_FORMAT_EXE:FILEPATH=CLANG_FORMAT_EXE-NOTFOUND
CMAKE_BUILD_TYPE:STRING=Debug
CMAKE_INSTALL_PREFIX:PATH=/usr/local
UA_ARCHITECTURE:STRING=posix
UA_BUILD_EXAMPLES:BOOL=ON
UA_BUILD_TOOLS:BOOL=ON
UA_BUILD_UNIT_TESTS:BOOL=OFF
UA_ENABLE_AMALGAMATION:BOOL=OFF
UA_ENABLE_COVERAGE:BOOL=OFF
UA_ENABLE_DISCOVERY:BOOL=ON
UA_ENABLE_DISCOVERY_MULTICAST:BOOL=OFF
UA_ENABLE_ENCRYPTION:BOOL=OFF
UA_ENABLE_EXPERIMENTAL_HISTORIZING:BOOL=OFF
UA_ENABLE_HISTORIZING:BOOL=OFF
UA_ENABLE_METHODCALLS:BOOL=ON
UA_ENABLE_NODEMANAGEMENT:BOOL=ON
UA_ENABLE_QUERY:BOOL=OFF
UA_ENABLE_STATIC_ANALYZER:STRING=OFF
UA_ENABLE_SUBSCRIPTIONS:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS_EVENTS:BOOL=OFF
UA_LOGLEVEL:STRING=300
UA_NAMESPACE_ZERO:STRING=FULL

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): master 21e32dd9973096133a2fcf826809518e5fd0bdfc
[NodeIdNotPersist.pdf](https://github.com/open62541/open62541/files/3370047/NodeIdNotPersist.pdf)

 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Ubuntu 18.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached This being seen in example program 'server_inheritance' as well as custom application
 - [x] Critical issue

GeorgeSellner, 2019-07-08:
I went back and did a build with the 0.3 7fb7c222b3be5c3948f91f14ebf5874e9463d57a
This was not an issue in 0.3

GeorgeSellner, 2019-07-08:
### OPC-UA Standard:

[https://opcfoundation.github.io/UA-TypeRepository/Core/docs/Part3/](https://opcfoundation.github.io/UA-TypeRepository/Core/docs/Part3/)

> **5.2.2 NodeId (link)**
>
> Nodes are unambiguously identified using a constructed identifier called the NodeId. Some Servers may accept alternative NodeIds in addition to the canonical NodeId represented in this Attribute. **A Server shall persist the NodeId of a Node, that is, it shall not generate new NodeIds when rebooting.** The structure of the NodeId is defined in 8.2.

jpfr, 2019-07-08:
I agree. Nodeids should be stable.

Somehow, the nodeid you look at is randomly generated.

We had deterministic initialization of the RNG in the past.
https://github.com/open62541/open62541/blob/master/src/server/ua_server.c#L225
But that lead to repeated session tokens and problems in different areas.

My suggestion is we go, find the source where randomness is added here and make it deterministic.

miehe-dup, 2019-08-15:
Any progress on this?
Related questions: 
* Can I change the nodeid of a node at runtime? 
* Can I set the nodeid of an automatically instantiated child node? … (Obviously) only relevant for nodes created from objecttypes or variabletypes, e.g. refer to tutorial_server_object.c

GeorgeSellner, 2020-10-05:
Resolved issue for my project by UA_ENABLE_DETERMINISTIC_RNG = on in cmake
also see 

    UA_GlobalNodeLifecycle lifecycle;

    lifecycle.generateChildNodeId = customChildNodeId;	/* Set callback so generate NodeIds using our custom function  */

GeorgeSellner, 2020-10-05:
I think this has been addressed.

closed 2020-10-05 by GeorgeSellner

ranote-009, 2025-01-16:
> Resolved issue for my project by UA_ENABLE_DETERMINISTIC_RNG = on in cmake also see
> 
> ```
> UA_GlobalNodeLifecycle lifecycle;
> 
> lifecycle.generateChildNodeId = customChildNodeId;	/* Set callback so generate NodeIds using our custom function  */
> ```

Hi @GeorgeSellner,

Thank you for sharing your solution! The approach of using UA_ENABLE_DETERMINISTIC_RNG and the lifecycle.generateChildNodeId callback is very insightful.

Would you mind sharing the implementation of your custom customChildNodeId function? Additionally, it would be great if you could outline the steps you followed to ensure NodeIds persist across server restarts. This would be incredibly helpful for understanding and applying a similar strategy in my project.