open62541/open62541 #2875: Browsing published nodes for node-id

dferrel71, 2019-07-09:
## Description
My server application adds two nodes for publishing two temperature values. I add the first temperature node like follows. The second temperature node is added in the same way but with the nodeid {1; 1002} and the name temperature2. 

    void addSensorInstance(UA_Server* server){
        UA_NodeId myNodeId = UA_NODEID_NUMERIC(1, 1001);
        UA_ObjectAttributes oAttr = UA_ObjectAttributes_default;
        oAttr.displayName = UA_LOCALIZEDTEXT("en-US", temperature1);
        UA_Server_addObjectNode(server, myNodeId,
                            UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
                            UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES),
                            UA_QUALIFIEDNAME(1, temperature1),
                            baseSensorTypeId, /* this refers to the object type
                                           identifier */
                            oAttr, NULL, NULL);
    }

As you can see in the code above, both temperature nodes are derived from an object type node which defines a variable node for the sensor value and the sensor name. 
If I connect to this server via UAExpert and look through these variable nodes, I found that the server has assigned them a nodeid too. These nodeid starts around 50220: The first temperature1-value-node has the numeric identifier 50220 and the second one has 50224.

Now to my questions: 
How can I assign my own nodeids to the variable nodes when instantiating my sensor instances? Why are the given variable nodeids 4 apart of each other? Is there a reason why the variable nodeids start at 50220?
In order to generically add ValueCallbacks to my variable nodes (to update the temperature value), I need to know these nodeids at runtime. I guess I can get them by browsing the variable nodes? How is this done?

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v0.3.0
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Ubuntu Linux

johndzilva, 2019-07-11:
https://github.com/open62541/open62541/issues/2818
Have a look at this one

teccheck, 2025-02-25:
The linked issue (#2818) seems to explain it quite well, so I'll close this one

closed 2025-02-25 by teccheck