open62541/open62541 #2876: Error binding a server socket

czqinyes, 2019-07-10:
when i start a server , error tip:Error binding a server socket
[image]

schroeder-, 2019-07-10:
Have you tried to start with an other Port like 4880?

johndzilva, 2019-07-10:
Your system may have some background application running in port 4840 (LDS).

Change this line in your server code to run your application in some other port 

`UA_ServerConfig_setMinimal(config, 4801, NULL);`

czqinyes, 2019-07-10:
> Your system may have some background application running in port 4840 (LDS).
> 
> Change this line in your server code to run your application in some other port
> 
> `UA_ServerConfig_setMinimal(config, 4801, NULL);`

Thanks a lot , i have taskkill the progress, and go well

ccvca, 2019-07-13:
@czqinyes If this issue is solved, please close.

closed 2019-07-15 by czqinyes