open62541/open62541 #2877: Cimpiling server under windows 10

mrburen121, 2019-07-10:
## Description
I'm unable to compile the sample server application under windows 10 with the amalgamation file.

## Background Information / Reproduction Steps
Installed:
MinGw
LLVM (clang)

`C:\Cmake\bin\cmake-gui.exe .. -G "Visual Studio 16 2019"`
Cmake logout
```
Used CMake options:
-DUA_ENABLE_HISTORIZING:BOOL="1" -DUA_ENABLE_PUBSUB_DELTAFRAMES:BOOL="1" -DUA_ENABLE_PUBSUB_CUSTOM_PUBLISH_HANDLING:BOOL="0" -DUA_ENABLE_PUBSUB:BOOL="1" -DUA_ENABLE_PUBSUB_INFORMATIONMODEL:BOOL="1" -DUA_ENABLE_PUBSUB_INFORMATIONMODEL_METHODS:BOOL="1" -DUA_ENABLE_AMALGAMATION:BOOL="1" -DCMAKE_CONFIGURATION_TYPES:STRING="RelWithDebInfo" -DUA_ARCHITECTURE:STRING="win32" 

Cache file:
UA_ENABLE_HISTORIZING:BOOL=1
UA_ENABLE_PUBSUB_DELTAFRAMES:BOOL=1
UA_ENABLE_PUBSUB_CUSTOM_PUBLISH_HANDLING:BOOL=0
UA_ENABLE_PUBSUB:BOOL=1
UA_ENABLE_PUBSUB_INFORMATIONMODEL:BOOL=1
UA_ENABLE_PUBSUB_INFORMATIONMODEL_METHODS:BOOL=1
UA_ENABLE_AMALGAMATION:BOOL=1
CMAKE_CONFIGURATION_TYPES:STRING=RelWithDebInfo
UA_ARCHITECTURE:STRING=win32
```
```bash
gcc -std=c99 -DUA_ARCHITECTURE_WIN32 server.c open62541.c -o server -lws2_32
```
and
```bash
gcc -std=c99 -DUA_ARCHITECTURE_WIN32 server.c open62541.c -o server`
```
Give this output:
```
C:\Users\cketterink\Desktop\V1.4\testserver>gcc -std=c99 -DUA_ARCHITECTURE_WIN32 server.c open62541.c -o server
In file included from open62541.c:28:
open62541.c: In functie Ô01mUA_NodeId_toStringopen62541.h:316:3: let op: implicit declaration of function Ô01m_snprintf_sÔid you mean Ô01m_swprintf_pÔ-Wimplicit-function-declaration]
   _snprintf_s(source, size, _TRUNCATE, __VA_ARGS__)
   ^~~~~~~~~~~
open62541.c:21414:19: note: in expansion of macro Ô01mUA_snprintfÔnprintfLen = UA_snprintf(nsStr, 10, "ns=%d;", nodeId->namespaceIndex);
                   ^~~~~~~~~~~
open62541.h:316:29: fout: Ô01m_TRUNCATEÔndeclared (first use in this function); did you mean Ô01m_RGNDATAÔsnprintf_s(source, size, _TRUNCATE, __VA_ARGS__)
                             ^~~~~~~~~
open62541.c:21414:19: note: in expansion of macro Ô01mUA_snprintfÔnprintfLen = UA_snprintf(nsStr, 10, "ns=%d;", nodeId->namespaceIndex);
                   ^~~~~~~~~~~
open62541.h:316:29: note: each undeclared identifier is reported only once for each function it appears in
   _snprintf_s(source, size, _TRUNCATE, __VA_ARGS__)
                             ^~~~~~~~~
open62541.c:21414:19: note: in expansion of macro Ô01mUA_snprintfÔnprintfLen = UA_snprintf(nsStr, 10, "ns=%d;", nodeId->namespaceIndex);
                   ^~~~~~~~~~~
open62541.c: In functie Ô01mUA_Server_DataSetReader_addTargetVariablesopen62541.h:316:29: fout: Ô01m_TRUNCATEÔndeclared (first use in this function); did you mean Ô01m_RGNDATAÔsnprintf_s(source, size, _TRUNCATE, __VA_ARGS__)
                             ^~~~~~~~~
open62541.c:30846:9: note: in expansion of macro Ô01mUA_snprintfÔ01;36mUA_snprintf(szTmpName, sizeof(szTmpName), "%s",
         ^~~~~~~~~~~
In file included from open62541.c:28:
open62541.c: In functie Ô01mUA_PubSubChannelUDPMC_openopen62541.h:290:22: let op: implicit declaration of function Ô01mInetPtonÔid you mean Ô01minet_ntoaÔ-Wimplicit-function-declaration]
 #define UA_inet_pton InetPton
                      ^~~~~~~~
open62541.c:64837:5: note: in expansion of macro Ô01mUA_inet_ptonÔ01;36mUA_inet_pton(AF_INET, addressAsChar, &imr_interface);
     ^~~~~~~~~~~~
In file included from open62541.c:28:
open62541.h:294:27: let op: implicit declaration of function Ô01mif_nametoindexÔid you mean Ô01mUA_if_nametoindexÔ-Wimplicit-function-declaration]
 #define UA_if_nametoindex if_nametoindex
                           ^~~~~~~~~~~~~~
open62541.c:64960:37: note: in expansion of macro Ô01mUA_if_nametoindexÔroup.ipv6.ipv6mr_interface = UA_if_nametoindex(interfaceAsChar);
                                     ^~~~~~~~~~~~~~~~~
open62541.c: In functie Ô01mUA_DateTime_localTimeUtcOffsetopen62541.c:65275:3: let op: implicit declaration of function Ô01mgmtime_sÔid you mean Ô01mgmtimeÔ-Wimplicit-function-declaration]
   gmtime_s(&ptm, &rawtime);
   ^~~~~~~~
   gmtime
In file included from open62541.c:28:
open62541.c: In functie Ô01maddServerSocketopen62541.c:65605:44: fout: Ô01mIPV6_V6ONLYÔndeclared (first use in this function)
       UA_setsockopt(newsock, IPPROTO_IPV6, IPV6_V6ONLY, (const char *)&optval,
                                            ^~~~~~~~~~~
open62541.h:286:29: note: in definition of macro Ô01mUA_setsockoptÔetsockopt(sockfd, level, optname, (const char *)(optval), optlen)
                             ^~~~~~~
In file included from open62541.c:28:
open62541.c: In functie Ô01mServerNetworkLayerTCP_startopen62541.h:316:29: fout: Ô01m_TRUNCATEÔndeclared (first use in this function); did you mean Ô01m_RGNDATAÔsnprintf_s(source, size, _TRUNCATE, __VA_ARGS__)
                             ^~~~~~~~~
open62541.c:65668:3: note: in expansion of macro Ô01mUA_snprintfÔ01;36mUA_snprintf(portno, 6, "%d", layer->port);
   ^~~~~~~~~~~
open62541.c: In functie Ô01mUA_ClientConnectionTCP_initopen62541.h:316:29: fout: Ô01m_TRUNCATEÔndeclared (first use in this function); did you mean Ô01m_RGNDATAÔsnprintf_s(source, size, _TRUNCATE, __VA_ARGS__)
                             ^~~~~~~~~
open62541.c:66121:3: note: in expansion of macro Ô01mUA_snprintfÔ01;36mUA_snprintf(portStr, 6, "%d", port);
   ^~~~~~~~~~~
open62541.c: In functie Ô01mUA_ClientConnectionTCPopen62541.h:316:29: fout: Ô01m_TRUNCATEÔndeclared (first use in this function); did you mean Ô01m_RGNDATAÔsnprintf_s(source, size, _TRUNCATE, __VA_ARGS__)
                             ^~~~~~~~~
open62541.c:66180:3: note: in expansion of macro Ô01mUA_snprintfÔ01;36mUA_snprintf(portStr, 6, "%d", port);
```

Sample server code from here: [https://github.com/open62541/open62541](https://github.com/open62541/open62541)
```

#include "open62541.h"

UA_Boolean running = true;
void signalHandler(int sig) {
    running = false;
}

int main(int argc, char** argv)
{
//    signal(SIGINT, signalHandler); /* catch ctrl-c */

    /* Create a server listening on port 4840 */
    UA_Server *server = UA_Server_new();
    UA_ServerConfig_setDefault(UA_Server_getConfig(server));

    /* Add a variable node */
    /* 1) Define the node attributes */
    UA_VariableAttributes attr = UA_VariableAttributes_default;
    attr.displayName = UA_LOCALIZEDTEXT("en-US", "the answer");
    UA_Int32 myInteger = 42;
    UA_Variant_setScalar(&attr.value, &myInteger, &UA_TYPES[UA_TYPES_INT32]);

    /* 2) Define where the node shall be added with which browsename */
    UA_NodeId newNodeId = UA_NODEID_STRING(1, "the.answer");
    UA_NodeId parentNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);
    UA_NodeId parentReferenceNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES);
    UA_NodeId variableType = UA_NODEID_NULL; /* take the default variable type */
    UA_QualifiedName browseName = UA_QUALIFIEDNAME(1, "the answer");

    /* 3) Add the node */
    UA_Server_addVariableNode(server, newNodeId, parentNodeId, parentReferenceNodeId,
                              browseName, variableType, attr, NULL, NULL);

    /* Run the server loop */
    UA_StatusCode status = UA_Server_run(server, &running);

    UA_Server_delete(server);
    return status;
}
```

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): 0.4.0 -dev (lastest version
 - [ ] Other OPC UA SDKs used (client or server): 
 - [X] Operating system: Windows 10 
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [X] Self-contained code example attached
 - [ ] Critical issue

Pro, 2019-07-23:
Duplicate of #1734

Try to use the mingw64 package, that should work. Mingw32 is missing some features
https://mingw-w64.org/doku.php

Or msys2
https://www.msys2.org/

Please reopen if necessary.

closed 2019-07-23 by Pro

EvgKlin, 2021-05-27:
@mrburen121 Do you find solution?

mrburen121, 2021-05-31:
No, I didn't found a solution.

EvgKlin, 2021-06-01:
@mrburen121 see this issue https://github.com/open62541/open62541/issues/1580.